dalathegreat/Battery-Emulator #315: Huawei inverter

soflomat, 2024-05-24:
Hello @dalathegreat,

Awesome work on the battery emulator. I just finished installing a Huawei 6kW L1 with 22 panels (11 on each string) ([Dmeg 425W](https://www.dmegcsolar.com/upload/img/2024-01/6596691717ab6.pdf)).

I wanted to clarify if the solar PV voltage requirement (below 600V) refers to the sum of both strings or each string individually.

Will the Dala Emulator work in my case? And which battery would you recommend?

Thank you for your help.

Best regards,

dalathegreat, 2024-05-25:
@soflomat , unfortunately the Huawei is most likely a dead-end for most people, see
https://github.com/dalathegreat/Battery-Emulator/issues/15#issuecomment-1932718008

closed 2024-05-25 by dalathegreat

soflomat, 2024-05-26:
I dont understand why Huawei is on the list of compatible inverter.
What work ?
How i can help ? Or finance to help to become compatible ?

Or perhaps, a emulator between luna2000 and battery ?

Thanks for your help

spelzl, 2024-05-26:
The only remaining way forward is to 'hack'/re-engineer the internal Luna2000 battery module communications and connect the EV battery to the Luna2000-C0 BMS Control Module.

I haven't started the process and I won't get to it shortly. So any active support is welcomed and appreciated.

soflomat, 2024-05-27:
What kind of active support would you like?

spelzl, 2024-05-27:
@soflomat Do you own a Luna2000 ESS?
If so, are you familiar with ModBus and do you have a sniffing tool and programme to listen to the bus communication and take logs?

peternooy, 2024-08-31:
@spelzl I agree with your approach and will seek a reference site to gather information as per you suggested. But there used to be a log of (Sun2000 10-KTL-M1 + Luna2000-5-S0 5kWh): shared by Dalah at https://we.tl/t-HNMUSylvey, but it is expired. Is this still available somewhere, how can I get it?