dalathegreat/Battery-Emulator #336: Github actions fail due to ESP32 randomness

dalathegreat, 2024-06-05:
The Github Workflows is failing randomly, due to the workers sometimes using ESP2.0.16 , and sometimes they use 3.0.0

We migrated the code to make it >3.0.0 compatible only, so it would be good to get the actions locked down to this version. I tried to make it work, but I could not figure out how to do it. 

@lenvm , any ideas? 😅

lenvm, 2024-06-05:
Hi @dalathegreat,

Can you please provide links to the specific workflow run numbers that fail “randomly”? This will help me debugging.

Thanks!

dalathegreat, 2024-06-05:
@lenvm Yes, these three workflows are affected:
- Compile All Batteries
- Compile All Inverters
- Compile All Combinations

Here is an example of failure;
https://github.com/dalathegreat/Battery-Emulator/actions/runs/9370957041/job/25799129523

When it uses 2.0.16
[image]

lenvm, 2024-06-05:
What different things have you already tried to make it work?

dalathegreat, 2024-06-06:
@lenvm , yes I tried this with 3.0.0, also retested with 3.0.1 now when it became available:
[image]
https://github.com/dalathegreat/Battery-Emulator/actions/runs/9399469733/job/25886951954

I also tried to add 
`arduino-platform: esp32:esp32@3.0.0 # latest 3.0.0 version as of 5/2024` but this also failed

dalathegreat, 2024-06-06:
The most promising way would be this;
```
arduino-cli core update-index --additional-urls https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json
arduino-cli core install esp32:esp32@3.0.1
```

It tries to download the list which has esp32 3.0.1 in it, but the build still fails https://github.com/dalathegreat/Battery-Emulator/actions/runs/9399822285/job/25888066755

dalathegreat, 2024-06-16:
CLosing this, Github seems to have resolved its workers!

closed 2024-06-16 by dalathegreat