dalathegreat/Battery-Emulator #338: Feature request : Add Dacia Spring Battery support

PaulBERDIEL, 2024-06-07:
Hello, I found this very interesting project and it is all I need ! Well done guys your code is very cool !
I am currently searching how to connect a Dacia Spring battery to my fronius Gen24. 
If someone have information about can protocol of this battery or DBC File I am very interested. 
I could work to add a full support of this battery pack in this project if I have the good informations.

Thanks by advance 

Paul

jakub8109, 2025-02-08:
I'm interested too

MathieuSdu02, 2025-02-10:
Hello, I'm Mathieu and I have the same project. I'm from north of France (Aisne - near Guise / Saint-Quentin). I own a Spring battery (from a disassembled one after accident) and I plan to buy a Fronius Primo GEN24 plus... First step to work on a communication module between Fronius and my battery BMS and after try to apply same to a normal car (to get bidirectional charging). Where are you and what is the status of your project ?

dalathegreat, 2025-03-15:
Dacia Spring integration: https://github.com/dalathegreat/Battery-Emulator/pull/971

Wiki, in progress: https://github.com/dalathegreat/Battery-Emulator/wiki/Battery:-Dacia-Spring-%E2%80%90-Renault-K%E2%80%90ZE

PR #971 (2025-04-13): New battery: Add support for CMFA-EV battery
### What
This PR implements support for the CMFA-EV platform

### Why
To add support for [Dacia Spring / Renault K‐ZE batteries ](https://github.com/dalathegreat/Battery-Emulator/wiki/Battery:-Dacia-Spring-%E2%80%90-Renault-K%E2%80%90ZE). User requested feature, fixes #441 and fixes #338

### How
Info based on the log files provided by Fbostrom on the Discord server :raised_hands:
closed 2025-04-13 by dalathegreat