dalathegreat/Battery-Emulator #348: Integration SAFT Batterie

SN00CS, 2024-06-15:
Hi,
we are about 20 persons, who have an early ESS, which has a SAFT Lithium-Ion Battery System with 10 Synerion Moduls (max. 560V 25kWh) inside. A similar system is build in a BOSCH ESS, but with only max. 6 Moduls. 
Is there anybody, who can program/integrate the battery in this project? There is no revers engineering needed, because the whole CAN registry is known. Hope someone can help us to integrate it to Battery-Emulator.

dalathegreat, 2024-06-15:
Hi @SN00CS 
I can code in support for this battery type, please email me (check my profile) and we can take it from there!

dalathegreat, 2024-06-15:
Turns out the document is a CANopen document, not something that easily integrates with this project unfortunately :/

dalathegreat, 2024-12-29:
Closing as wont do at the moment, due to no CANopen implementation exists

closed 2024-12-29 by dalathegreat