dalathegreat/Battery-Emulator #353: BMW i3 battery not working

DeadmaroZ-TLOTL, 2024-06-18:
When i connect battery in main screen show System status: FAULT, but in Events nothing, when i disconnect pin 9 in events show battery empty and not can communication. What am I doing wrong?
[image]
[image]
[image]

odlg, 2024-06-18:
That is not a lot of information. Have you by accident mirrored the pins? Do you have a terminating resistor on the can-bus? How long are the cables? Are the cables properly seated?

DeadmaroZ-TLOTL, 2024-06-19:
No, pins not mirrored. Cable 20cm, all seated properly! Need another resistor on the CAN line? If so, who and where?

DeadmaroZ-TLOTL, 2024-06-19:
I realized what I'm doing wrong, no HV Plug connected! Can't the interlock be bypassed somehow?

odlg, 2024-06-19:
A 120Ω resistor over the CAN bus will probably make it more stable = fewer errors in the long run. 
You need to short the pins covered by the orange slider on HV port - I do not know of other options. But CAN communication should work nonetheless.
What amps are the battery drawing over 12V? Should be around 0.4 A.

DeadmaroZ-TLOTL, 2024-06-19:
Values sent to inverter: Real SOC%: 61.10 Battery voltage: 387.50 Battery current: 0.00 Wh when full: 30000 Remaining Wh: 22000 Max charge power: 0 Max discharge power: 196596 Active power: 0 Min temp: 18.00 Max temp: 18.00ERROR: SOC% reported by battery not plausible. Restart battery!
CAN message failed to send within defined time. Contact developers, CPU load might be too high.
Info: Battery failed to precharge. Check that capacitor is seated on high voltage output.

DeadmaroZ-TLOTL, 2024-06-19:
Another battery show this, but switching relays.

[image]
[image]

Values sent to inverter: Real SOC%: 58.90 Battery voltage: 383.00 Battery current: 0.00 Wh when full: 30000 Remaining Wh: 22000 Max charge power: 0 Max discharge power: 196596 Active power: 0 Min temp: 18.00 Max temp: 17.00ERROR: High amount of corrupted CAN messages detected. Check CAN wire shielding!
CAN message failed to send within defined time. Contact developers, CPU load might be too high.

odlg, 2024-06-19:
So, you have a battery that communicates and another one that does not?
Are the battery and the solax inverter on the same can-bus? That might introduce some issues.

dalathegreat, 2024-06-19:
HV plug needs to be seated, and the capacitor is also needed on the HV output!

DeadmaroZ-TLOTL, 2024-06-20:
> So, you have a battery that communicates and another one that does not? Are the battery and the solax inverter on the same can-bus? That might introduce some issues.

One battery show fault in main screen, but in events screen show nothinb, but another battery in main screen show OK, but in events show this >

[image]
[image]

DeadmaroZ-TLOTL, 2024-06-20:
> HV plug needs to be seated, and the capacitor is also needed on the HV output!

In HV  plug I short small plug pins and fault about interlock got lost. Could the can error be due to the unconnected capacitor?

odlg, 2024-06-21:
The precharge failure is caused by the missing capacitor. HV contactors won't close without it. 
It is recommended to twist the CAN cables to reduce noise, you could try that to avoid the CAN_RX_WARNING. Also, adding a 120Ω resistor across the can cables would probably help.
A single CAN_OVERRUN INFO message is normal.

DeadmaroZ-TLOTL, 2024-09-09:
No!

dalathegreat, 2024-11-04:
@DeadmaroZ-TLOTL the latest software releases contain a "More battery info" page for BMW i3, which can be used for troubleshooting:

[image]

dalathegreat, 2025-02-16:
Closing this one, we have over 20 people using i3, and have also improved the [wiki troubleshooting section](https://github.com/dalathegreat/Battery-Emulator/wiki/Battery:-BMW-i3#troubleshooting-tips) for i3. Incase you still have issues getting contactors to close, most likely the i3 BMS is crashed, and it has entered this locked state that requires EoS tester to unlock. Feel free to comment here again and we can re-open if needed!

closed 2025-02-16 by dalathegreat