dalathegreat/Battery-Emulator #356: Best set AC coupling Dala

soflomat, 2024-06-23:
Hello everyone,

I have a brand new Huawei installation with optimizers. I understand that the SUN2000 will not be compatible with DALA.
So I'm thinking of putting in a new AC inverter, just for the battery.
What is the best AC inverter and battery that you recommend? 
I'm thinking of putting it all together in August.

Thanks community

dalathegreat, 2024-06-24:
Hi @soflomat , selecting an inverter is highly dependent on your location, which ones are approved by your grid supplier, local availability etc.

I don't specifically recommend this inverter, but the Fronius Gen24 can be used without solarpanels, and set to charge the battery from other AC generators in your house. So this inverter is an option (albeit an expensive one...)

Good luck with the build!

soflomat, 2024-06-26:
I check SMA it's big case, but high Kw. What your sentiment ?

What is the best battery ans seller in Europe ?

@dalathegreat

dalathegreat, 2024-06-30:
@soflomat , think it is best if you put these questions on a discussion forum, like Secondlifestorage.com for instance. We mainly use the Github for issue tracking, not purchase advice.

I personally recommend getting batteries from GBD.no incase you are close to Norway. 

I will close this issue!

closed 2024-06-30 by dalathegreat