dalathegreat/Battery-Emulator #370: SMA SB SE

alex-military, 2024-07-07:
Hi, I see that you are implementing a tripower smart energy SMA but will it also be compatible with single-phase ones? i.e. with Sunny Boy Smart Energy?
Will this protocol be implemented with batrium BMS?

alex-military, 2024-07-08:
I worked for a year with SMA inverters and they work with BYD batteries but have an automatic recognition mode. I intend to create a system in my home with a sunny boy smart energy or a tripower smart energy to take advantage of full backup. If it is not very complicated to find these registers I will gladly help you if up to then you have not managed to implement them

dalathegreat, 2024-07-08:
The current SMA protocol implemented has wrong bits set in some places. We need to emulate a RESU battery, not BYD. Here is a screenshot from a Sunny Boy Storage inverter setup with the Battery-Emulator. It does not work

[image]

Neither does the SMA-Tripower protocol. We need CAN logs to proceed from a working system.

alex-military, 2024-07-09:
OK could you explain to me how I can extract or find these can registers? I still have a few contacts who install those inverters and maybe I can convince them to let me connect to the inverters and try to recover them but that's something I've never done

dalathegreat, 2024-07-09:
@alex-military , see this Wiki page I wrote on the subject of logging; https://github.com/dalathegreat/Battery-Emulator/wiki/Data-needed-for-new-inverter-integration

alex-military, 2024-07-10:
OK, let's see if I understood correctly because the translation doesn't help me. I use Google Translate to understand. in practice with a working system (sma tripower SE inverter or sunny boy SE and BYD battery) I connect a lilygo in parallel on the CAN cables and note down the data (even the time?) of %soc temp charge/discharge status temp etc etc. then by leaving the recording they must simulate sudden battery shutdown sudden shutdown inverter charge discharge and note down the event messages that appear on the inverter dashboard? finally, through a cross comparison, the various values ​​obtained in the register are compared to map them and, if necessary, all the rest of the data is interpreted. should it be right?

dalathegreat, 2024-07-10:
Yes. Most important is to get Startup log, and operation log.

This page has more info on how to log: https://github.com/dalathegreat/Battery-Emulator/wiki/CAN-logging

Once you have log files, post them here and we can fix the protocol!

dalathegreat, 2024-07-31:
@alex-military 
SMA-CAN has been improved in this commit, I got RESU logs that at the same time improved the BYD code

https://github.com/dalathegreat/Battery-Emulator/commit/dbab56944d1705ae825169fcb5b441a3c6c18c0a#diff-18b344f7c221f3d197c188c1d6700b37cb53e455243ca7b9298cc67a2b46497b

Please try dropping in this SMA-CAN.cpp file and restesting

alex-military, 2024-08-18:
Hi, I tried with some ex-colleagues if I could go and do the data recording but unfortunately no one gave me availability. I was thinking of doing an installation myself but not having the battery I don't know how I could help

dalathegreat, 2024-08-18:
Hi, We are working nighlty on adding SMA support on the [Patreon Discord](https://www.patreon.com/dala) server. We have CAN logs, and are working out the pairing process.

Nokius, 2024-09-14:
@dalathegreat are logs still needed from Storage Boys? Could log from 2.6 and 3.6 with BYD HVS and HVM.

dalathegreat, 2024-09-16:
@Nokius SMA Sunny Boy support was added a few weeks ago: https://github.com/dalathegreat/Battery-Emulator/wiki/SMA-Sunny-Boy-Storage-inverters

alex-military, 2024-09-28:
Hi, I saw that support for sunny boy storage has been added, but does this protocol also work with sunny boy smart energy and sunny tripower smart energy? Or the sma hybrid inverters? I should buy some soon but I was undecided due to the cost of the batteries between the sunny boy storage and the hybrid deye.

dalathegreat, 2024-09-28:
@alex-military Most likely they do not, so I'd advise waiting until someone can provide logs from a Tripower / SBSE setup

KoenM1, 2024-10-30:
Hi, 
I have an SMA tripower smart energy hybride converter. 
I did by the Batrium BMS for a 350v battery.  
Now it seems that the BMS is not communicating with the inverter. 
Did you find out the can communication?

PR #641 (2024-12-28): Bugfix: Make SMA Tripower work
### What
This PR fixes the SMA Tripower protocol implementation

### Why
To allow for SMA Tripower inverters to be used with the project

### How
Testing performed on the Discord server

Fixes #370
closed 2024-12-28 by dalathegreat