dalathegreat/Battery-Emulator #377: Renault ZOE 25.2 kw battery to the inverter Deye SUN-15K-SGO1HP3-EU-AM2

valentzoe, 2024-07-11:
Hi, dalathegreat! I have been trying for quite a long time to connect the Renault ZOE 25.2 kw battery to the inverter Deye SUN-15K-SGO1HP3-EU-AM2. I think I have done all the settings in the three files according to your recommendations.  Separately checked the CAN signals from the battery and the inverter. In both cases LilyGO responded with flashing green indicator. When trying to communicate with the battery the error F58" BMS_Communication_Fault" appeared. I tried to fulfil all the recommendations related to this error. I'm need your help. Thanks a lot
[image]
[image]
[image]
[image]
[image]

dalathegreat, 2024-07-11:
Zoe support is still very untested. We have been discussing and testing things back and forth on the Discord server. Would be interesting to see if all values look OK in the webserver. Could you post all the values you see there (green section)?

Deye inverter guide is here: https://github.com/dalathegreat/Battery-Emulator/wiki/Deye-inverters

But it would be best to get the battery values confirmed working before we add the inverter into the mix!

valentzoe, 2024-07-12:
I thank you for your quick response. I am also interested in cooperating on this issue. I will provide further information shortly. Please note that the signal "2AD Traction battery relay control 1" and "2AC Traction battery precharging relay control" require a "-" connection.  
I have recorded the CAN_H and CAN_L oscillograms on the batteries.
I agree with you that the battery should be checked with the monitor first.   
In all connection attempts I did not see any indication that the SSR Relay is on.  The circuit of SSR Relay is similar to your circuit.
Best regards

dalathegreat, 2024-07-12:
SSRs need to be able to trigger with 3.0VDC voltage level. Many cheap SSRs require 3.3V to function. We have listed models that work on the [wiki page](https://github.com/dalathegreat/Battery-Emulator/wiki/Contactor-Control-via-GPIO-pins)

I suggest leaving out the SSRs and manually turning on the battery to start with. Can you post all the values shown on the webserver, so we can see if the values make sense?

valentzoe, 2024-07-13:
It's not actually about the relay. Of course, I measured the voltage at the relay inputs - it was zero. I wanted to draw your attention to the fact that something is wrong. I also do it manually to turn on the battery.  The pictures show the information of the last connection.
[image]
[image]
[image]
[image]
[USER_SETTINGS_h.txt](https://github.com/user-attachments/files/16201475/USER_SETTINGS_h.txt)

dalathegreat, 2024-07-13:
Looks like no CAN messages are coming thru at all from the battery, all values are on default values.

Try checking that CAN-H and CAN-L aren't inverted accidentally!

valentzoe, 2024-07-13:
Good evening. No, they are not turned over. If they are turned over, then the LILyGO indicator lights up red. Mine lights up greenThanks.

dalathegreat, 2024-07-13:
Can you take a raw CAN log from the battery? See this page for info how: https://github.com/dalathegreat/Battery-Emulator/wiki/CAN-logging

valentzoe, 2024-07-16:
All attempts including “TEST_Fake_Battery” Canbus testing end up the same:
[image]
[image]
[image]
[image]

I tried versions 6.2 6.2.1 6.3. What I noticed.  If I do a “Verify” operation before “Uploud”, error messages appear.
[image]
[image]
[image]

Once again I would like to draw your attention that the control voltage never appears on I/O 25,32,33 when starting the program.
In your opinion, what do you think may be the problem lies in Lilygo, unlike your versions 1.0 I have version 1.1.

dalathegreat, 2024-07-16:
The control voltage to IO 25,32,33 wont be applied. Battery control is blocking it:
[image]

Both checkmarks need to be green for automatic control to work (Battery :heavy_check_mark:  Inverter :heavy_check_mark: )

What we need is raw CAN logs from your battery, so we can see what goes wrong with value mapping. If no values appear, you have connected the CAN wiring wrong. I**t looks like all your problems come from no CAN data from battery.**

Use "TEST_FAKE_BATTERY", and in the serial you will see massive amounts of CAN messages being printed (Ofcourse wiring needs to be correct)

valentzoe, 2024-08-04:
Good afternoon!

Did the following work.
It cost me a lot of effort to make sure that the battery from the Renault ZOE car functions without problems.
In a professional workshop the battery was installed on RENAULT ZOE Q210.
It was checked its performance, the ability to charge and discharge.
The battery showed excellent results SOH -96%, then it was dismantled for use as Second-Life-Battery for PV. 
[image]

[image]

[image]

[image]

Lilygo ver. 6.3 mit "Test Fake Battery" the result was extremely negative, no change.

I decided to test the LilyGo module in another way, namely I made an adapter for testing LilyGo and battery with CanZEplus.
And, surprisingly, with the help of CanZEplus and LilyGO the data at address 7BB, which are in the LilyGO memory, were read without errors, indicating all the parameters of the battery SOH, 96 Celle, as well as the Firmware version, which also corresponds to the tests carried out in the installed state on the car.

[image]
In my opinion battery testing with CanZEplus is worthy of attention.

Please note that it is not about Cap_H and CAN_L, the connection is done correctly, but the data received via Canbus in LilyGO memory are not correctly emulated in BYD and are not reflected in FAKE Battery and Serial Monitor

I worked with the latest version 6.4 and ESP32 ver 3.03. The installation of the program went without errors.

[image]

Please help me, without your help I can not solve this problem.

DIYBMSInverter, 2025-01-07:
Hey valentzoe,

Do you still have the issue? We are currently working on a DEYE 50kW inverter as well and face the F58 error message,

Kind regards

dalathegreat, 2025-01-15:
Closing this one as completed, many Zoe Gen1 users on the Discord now, and batteries work great! :+1:

closed 2025-01-15 by dalathegreat