dalathegreat/Battery-Emulator #383: Fronius Gen24 10KW 11Kwh BYD

Mights001, 2024-07-16:
Hello Guys, 

I have a BYD Battery and now want to expand it with another HV memory that communicates via Pylontech HV. Does that work with that? 
And how is the wiring and what do I need for it?

In the software I can only activate one Battery, it somehow works?

Thx. for reply

dalathegreat, 2024-07-16:
Most likely this wont work. Please add a diagram of what you intend to connect, and we can say if it will work or not!

Mights001, 2024-07-17:
Hello, here is a block diagram.
My first thought is to connect the two batteries together on the DC side, of course with the same voltage level.
The battery emulator should then query both batteries and pass them on to the Fronius via the BYD Modbus interface.

[image]

dalathegreat, 2024-07-18:
This is not supported, we have no way to add a BYD battery together with a Pylon battery unfortunately. But the software is open source, so you can attempt adding support for this specific combo!

closed 2024-12-12 by lenvm