dalathegreat/Battery-Emulator #387: KIA EV6 / E-GMP platform integration

pazdzierzz, 2024-07-19:
Thanks to a discord user **Sam79** we have can logs from KIA EV6 that we can work on.

**Full log of G Can and M Can from Car:**
[Kia_EV6GT_G-CAN_M-CAN.zip](https://github.com/user-attachments/files/16309486/Kia_EV6GT_G-CAN_M-CAN.zip)

**This file is used to test activate/deactivate the relays, it works fine.** 
[Kia_UDS_contactors (Contactor Test works fine).csv](https://github.com/user-attachments/files/16309489/Kia_UDS_contactors.Contactor.Test.works.fine.csv)

[https://github.com/dalathegreat/Battery-Emulator/wiki/Kia-EV6-battery](url)

Lisanderl, 2024-07-20:
Hello! I have an Ioniq 5 77 kw battery. I tried connecting it yesterday by this manual https://github.com/dalathegreat/Battery-Emulator/wiki/Kia-EV6-battery, but it didn't work and shows a CAN communication error. As I understand, integration is still in progress. Can I help with that?

Lisanderl, 2024-07-21:
> Hello! I have an Ioniq 5 77 kw battery. I tried connecting it yesterday by this manual https://github.com/dalathegreat/Battery-Emulator/wiki/Kia-EV6-battery, but it didn't work and shows a CAN communication error. As I understand, integration is still in progress. Can I help with that?

I probably had some issues with wires, but now it works and shows data from the battery. Can I try to connect it to some inverter ?

pazdzierzz, 2024-07-22:
> I probably had some issues with wires, but now it works and shows data from the battery. Can I try to connect it to some inverter ?

Sure! It'd be nice if you put some screenshots from webserver

JorgenSeemann, 2024-07-26:
> I probably had some issues with wires, but now it works and shows data from the battery. Can I try to connect it to some inverter ?

You can, but note that SOC doesn't seems to be updated with current setup. 
Some modification will allow you to close contractors manually (so rest of data gets updated)

JorgenSeemann, 2024-07-26:
> Thanks to a discord user **Sam79** we have can logs from KIA EV6 that we can work on.
> 
> **Full log of G Can and M Can from Car:** [Kia_EV6GT_G-CAN_M-CAN.zip](https://github.com/user-attachments/files/16309486/Kia_EV6GT_G-CAN_M-CAN.zip)
> 
> **This file is used to test activate/deactivate the relays, it works fine.** [Kia_UDS_contactors (Contactor Test works fine).csv](https://github.com/user-attachments/files/16309489/Kia_UDS_contactors.Contactor.Test.works.fine.csv)
> 
> User **JoSee68** sent CCS logs too:
> 
> **EV6** [decoded_ev6_data_full.zip](https://github.com/user-attachments/files/16309784/decoded_ev6_data_full.zip)
> 
> **Ioniq5** [IoniqMF4.zip](https://github.com/user-attachments/files/16309799/IoniqMF4.zip)
> 
> [https://github.com/dalathegreat/Battery-Emulator/wiki/Kia-EV6-battery](url)

Would you have any chance to narrow down the commands used for de/activating contractors ?

Lisanderl, 2024-08-02:
> > I probably had some issues with wires, but now it works and shows data from the battery. Can I try to connect it to some inverter ?
> 
> Sure! It'd be nice if you put some screenshots from webserver

[image]
[image]
[image]

Lisanderl, 2024-08-03:
I will try 'build with contactor', but how can I help implement regular relay control?

Lisanderl, 2024-08-21:
@pazdzierzz Hi! I found that the can-fd adapter requires some radiator cause it becomes pretty hot and fries itself.  I was trying to figure out why I got CAN FD init errors after a few minutes of usage. I have 3 broken boards now. 
Looks like it also damaged a BMS unit, because now, I can read any can fd data even with the new can-fd adapter.

StarkJohan, 2024-08-21:
> @pazdzierzz Hi! I found that the can-fd adapter requires some radiator cause it becomes pretty hot and fries itself. I was trying to figure out why I got CAN FD init errors after a few minutes of [...]

That will happen if you supply 5V on to the 3V3 input, ask me how I know...
But otherwise the controller shouldn't run too hot. It is indeed a bit warmer than other controllers (normal CAN controllers that I've compared with). My MCP2518FD has been running a few days and is currently at 17C above ambient temperature (44C) and it's specified temperature range is three times that.

havrla, 2024-10-27:
Have a nice day
I'll add to your point, the EV6 has several types of batteries, some have all the modules inside, some don't. Hard to say without disassembly. If the modules are not there, what you wrote will probably happen, some values ​​are missing.
[image]

Gerwaz3, 2024-11-22:
Hello, I have a battery like the one in the picture above, 180 cells 72kWh. 
Does the current version 7.8 show the correct SOC and SOH values? 
For me, SOH=100% SOC=3.5% U=614.6V Cell max=3400 Cell min=3400 Temp=5C. 
I measured the voltages with a meter and they match. It would be good to charge it a bit, but I have a lot of work to mount it in the right place and I would like to be sure that the SOH is still at a good level - the battery is from 2021. 
If any tests of this version of the battery are needed - I can help.

ilia92, 2025-01-11:
Hello @pazdzierzz 
Since the integration is done and the contactors are closing, do you think we're ready to close this issue ?

We have another issue with the charging limits, but I think this will require a new issue.

pazdzierzz, 2025-01-11:
Sure, good idea. Issue closed. 

> Hello @pazdzierzz Since the integration is done and the contactors are closing, do you think we're ready to close this issue ?
> 
> We have another issue with the charging limits, but I think this will require a new issue.

closed 2025-01-11 by pazdzierzz