dalathegreat/Battery-Emulator #389: Reset Panic: Stack Smashing Protect Failure

bubble07, 2024-07-24:
Panic resets observed roughly between 8-24 hours up time. Seems unrelated to load (has happened at idle, when charging and when discharging).

There have also been other instances of resets and CAN-OVERFLOW, without panic being logged.  The frequency of reset was between 30 seconds uptime through to 8 hours Thursday to Saturday last week. But this has been far less frequent over the last few days (roughly once per day).

Note that the system had been stood for about 3 weeks unused prior to last Thursday.

Logs of both panic and no panic resets attached.

[Error Log_No_Panic (2).txt](https://github.com/user-attachments/files/16357386/Error.Log_No_Panic.2.txt)
[Panic_Error_Log_23_07_24.txt](https://github.com/user-attachments/files/16357329/Panic_Error_Log_23_07_24.txt)
[USER_SETTINGSh.txt](https://github.com/user-attachments/files/16357330/USER_SETTINGSh.txt)
[USER_SETTINGScpp.txt](https://github.com/user-attachments/files/16357331/USER_SETTINGScpp.txt)

Many thanks 🙂

bubble07, 2024-07-28:
Issue can be avoided by turning off MQTT option as recommended by the discord community.

bubble07, 2024-08-02:
One instance of panic reset this morning, even with MQTT disabled, unfortunately serial logger was not running.

bubble07, 2024-08-19:
Still seeing Stack smashing panic reset with MQTT disabled, uptime was a new record at 5 days before it happened this morning. Latest log attached
[Stack Smash 240819_after5daysuptime.txt](https://github.com/user-attachments/files/16664030/Stack.Smash.240819_after5daysuptime.txt)

dalathegreat, 2024-08-20:
Are you running "DEBUG_VIA_USB" all the time? This can also be a source of crashes

bubble07, 2024-08-20:
I think I previously had USB debugging disabled, but turned it on at the same time as turning MQTT off. I'll try turning debugging off as well and report back

bubble07, 2024-09-19:
One instance of panic reset this morning about 3 minutes after an overnight battery charge had completed, uptime at time of reset was around 10 days 20h (previous best was around 6 days). This was running version 7.1.1.
No crash log available as USB logging was disabled.
Webserver and MQTT were running.

Observations:
CAN overrun errors significantly reduced in steps after the CAN refactoring update and also after the update to disable EEPROM writing.
As expected and suggested by Dala above, reducing the number of CPU intensive options also reduced CAN overrun errors. However I would like to be able to see what the emulator is doing, so I'd ideally like MQTT and web server, but will see if MQTT updates in v7.2 cover enough to disable the web server

bubble07, 2024-10-04:
Great, thanks, sounds promising! 

Will do, I'm remote to the system at the moment, but will try over the weekend when I'm back and let you know.

bubble07, 2024-10-06:
@amarofarinha I'm now running 7.4.1. However I have had some hardware reliability difficulties with an MCP2515 which has now been replaced. I'll report back as confidence increases with uptime, but we might not know if the change is due to software improvements or replacing defective hardware.

bubble07, 2024-11-01:
Stack smashing issue did not reoccur with SW update, however inverter did stop sending CAN messages at random intervals (typically 12-48h apart). This was fixed by reversing the roles of the Liligo CAN and the MCP2515 CAN, over 17 days uptime without issue since that swap 👍
Thank you for the support @amarofarinha & @dalathegreat

closed 2024-11-01 by bubble07