dalathegreat/Battery-Emulator #390: Don’t close contractors

RusVl, 2024-07-25:
[image]
contactors do not close even with soldered jumpers. What  other options are there?

dalathegreat, 2024-07-25:
Apart from HVIL, you need to have
- Pyrofuse mounted
- 60/120 Ohm resistor on X98 (depending on RWD / AWD)
- Capacitor on HV output
- PCS installed inside battery

See the wiki, for a detailed guide: https://github.com/dalathegreat/Battery-Emulator/wiki/Tesla-Model-S-3-X-Y-battery#example-integration

RusVl, 2024-07-25:
[image]

RusVl, 2024-07-25:
I don’t have any idea!

RusVl, 2024-07-25:
[image]
I made all options. Only used resistor 2 Om and jumper instead pyro fuse.
[image]

RusVl, 2024-07-26:
[image]

RusVl, 2024-07-26:
May be I change bms of my battery in older bms?

dalathegreat, 2024-07-28:
Your screenshot shows low voltage (12V battery) as 7V. Looks like the power supply is not stable enough?

RusVl, 2024-08-01:
Дякую. Спробую іншу батарею.

RusVl, 2024-08-11:
the problem was solved by replacing the 22-year-old BMS with the 21-year-old BMS.
thanks to everyone who responded.

closed 2024-08-11 by RusVl