dalathegreat/Battery-Emulator #394: Tesla Model 3 contactors stuck in on off loop with Deye / Solark 30

DCWelsford, 2024-07-28:
I have been running a Tesla Model 3 battery on a Deye 30KW / Solark inverter. It works great, but it gets stuck in an on / off loop every 30 seconds. This loop usually begins anywhere from 4 hours to 24 hours of starting it up. I can’t seem to figure out what sends it into the loop. In the common fixes listed, it says to unplug it and start it up again. This usually fixes the problem temporarily, but then it starts again. Anyone else had this problem or know of a permanent fix?

dalathegreat, 2024-07-31:
Hi @DCWelsford , When you say on/off loop, do you mean that contactors open, or just that the inverter turns off? Common issues are CAN problems, please check the wiki: https://github.com/dalathegreat/Battery-Emulator/wiki#can-wiring-troubleshooting

DCWelsford, 2024-08-01:
Dala the Great,

Thanks for all the work you have done that I have benefited from. I am working on powering a large system for a hospital in Haiti. The Deye / Solark inverter that I was practicing with I had to send in to the school in Haiti, so I am waiting on getting another to try. The problem would be that the contactors in the battery would open for about 30 seconds, and then close for about 30 seconds… this would then cause the inverter to shut down for 30 seconds, go through it’s starting procedure, close the contactors in the inverter, and then 10 seconds later the inverter would open, and the loop would begin again. The only way to stop it was to pull all power and connectors from the model 3 battery. I could never force replicate it, and sometimes it would run fine for hours and hours before it randomly started into the loop. I will try the grounding, shielding, and the other items you said to try when I get my new inverters.
DC Welsford

closed 2025-01-02 by dalathegreat