dalathegreat/Battery-Emulator #395: Hyndai/kia 64kw

JKi92, 2024-07-29:
Hei 
akku on toiminut 3 viikkoa.
Heräte akusta oli virta loppunut.vaihdoin akun bms lähti päälle ja heti sammui.
Nyt battery emulator näyttää akku tyhjä / alijännite. Mistä lähtisin vikaa etsimään?
[image]
[image]

JKi92, 2024-07-29:
[image]

dalathegreat, 2024-07-31:
Please use english to reach wider target audience.

Looks like CAN communication is entirely down. We have seen users with fried CAN chips, due to incorrect grounding. See CAN troubleshooting tips here: https://github.com/dalathegreat/Battery-Emulator/wiki#can-wiring-troubleshooting

closed 2024-12-12 by lenvm