dalathegreat/Battery-Emulator #403: MINI BEV F56 Batterie

Heliwolf, 2024-08-08:
I would like to know if the SE14 battery from the MINI F56 BEV is also supported. I think it should work because it has the same SME (Storage Management Electronics) control unit as the BMW i3 battery. Do you have any experience with this?

dalathegreat, 2024-08-10:
Unfortunately we wont know until anyone tries it. If you have the battery and are willing to help with the integration, see this wiki page: https://github.com/dalathegreat/Battery-Emulator/wiki/Data-needed-for-new-battery-integration

grafitix, 2025-09-02:
I've been using a 32.6 kWh Mini (F56) battery for some time now, and it works perfectly.
It's comparable to a 94 Ah BMW i3 battery.

dalathegreat, 2025-09-03:
@grafitix That is awesosme! Do we need a separate Wiki page for Mini batteries, or are the connections the same as on BMW i3?

dalathegreat, 2025-09-07:
Mini Cooper Electric added under the BMW i3 page! https://github.com/dalathegreat/Battery-Emulator/wiki/Battery:-BMW-i3#supported-bmw-batteries

Jack-from-Poland, 2026-11-14:
@grafitix Hey, which hybrid inverter did you pair it with? Are you using a LilyGO board too?

grafitix, 2026-11-14:
Hi Jack!
I use GoodWe and Solax inverters.
Both work without any issues with the Mini (BMW) battery.
My personal favorite is definitely the Solax!
The GoodWe, however, is significantly cheaper.
Unfortunately, I have no experience with Solis.
I use a Lilygo T2CAN.

Jack-from-Poland, 2026-11-14:
@grafitix 
Thanks so much for getting back to me! I’m trying to decide which hybrid inverter to buy, and the more I read, the less sure I am which one to go for.

I was considering GoodWe too, but I’ve read that the latest firmware no longer works with the LilyGO. Am I right in thinking you no longer use the GoodWe, so you haven’t had a chance to test the latest firmware yourself? Are you still running the battery setup at all?

Could you share the exact model of the SolaX and GoodWe inverters you used? One last thing, if you don’t mind. Could you share a photo of the inside of your DC enclosure and a list of the components you used?

Thanks again for your help!

grafitix, 2026-11-14:
I have several systems running.
I use a GoodWe GW10K-ET and a Solax Hybrid X3 G4 15K (both with the latest firmware).
Issues with new firmware can always arise—regardless of the manufacturer.
However, the BE-Team (dala) always responds very quickly, and so far, they have consistently found solutions in no time!
In general, though, you should always exercise caution when it comes to firmware updates for inverters.
Take a look at the ready-made units available at https://shop.redispo.se.
If you want a solution that is both simple and perfect, I would highly recommend purchasing one of these pre-built units!

Jack-from-Poland, 2026-11-14:
@grafitix 
Thanks for getting back to me! Sorry, just one last question. Once the initial setup is done, can any of your inverters restart the system automatically? Say the solar panels stop producing power overnight and the battery runs down. Will the whole system start up again on its own when the sun comes back out, without you having to do anything manually?
Thank you!

grafitix, 2026-11-14:
Do you want to operate your system as an island (without a grid connection)?

Jack-from-Poland, 2026-11-14:
Yes, my setup will be completely off-grid. It'll just be solar panels, a hybrid inverter, and a battery pack from a MINI. There won't be any connection to the grid.

grafitix, 2026-11-14:
Unfortunately, I have very little experience with off-grid (island mode) operation.
In principle, these inverters are capable of a "black start."
However, you must ensure that the 12V supply (for the BMS and the Lilygo) remains powered.
You’ll need to experiment a bit with that.

Also, what will you do if the battery is empty and the sun isn’t shining?
Do you have a generator?

Jack-from-Poland, 2026-11-14:
Yes, I already have a generator, and the Solis supports a generator connection. I’ve seen people work around the lack of a dedicated generator input on other inverters by rectifying the generator’s AC output to DC and feeding it into one of the PV MPPT inputs.

What makes the Solis appealing to me is that it has AFCI and four MPPT trackers, so I’d have plenty of room to expand the system later and eventually build it out to meet my long-term needs. The price is the main thing holding me back, though, so I’m still looking for cheaper alternatives.

I did consider GoodWe, but I took it off my shortlist after reading about the firmware compatibility issues mentioned on the Battery-Emulator’s inverter compatibility page.