dalathegreat/Battery-Emulator #404: Manual override - CAN contactors

frasersimpson83, 2024-08-12:
Would it be possible to add a manual override for batteries with CAN based contactor controls. 

This is so that if we want to open contactors but leave everything else in place we can.   Currently for Tesla Model 3 batteries for instance we need to remove 12V connection to the BMS and contactors to open them which is fine, but if remote from the battery it would be nice to be able to open the contactors and disconnect the battery manually.

This setting should stay active even during a restart for safety purposes and then the user has to connect to the web interface and put it back into 'normal operation' where the contactors would close automatically.

This would be extremely helpful in multiple situations and I'm guessing not that hard to implement since we already have all the code there, it's just adding a manual over-ride switch somewhere.

dalathegreat, 2024-08-12:
Afaik adding a button would not help for Tesla, since we have no known way to open the contactors once the pack is going.

frasersimpson83, 2024-08-13:
But when the Lilygo errors out for whatever reason and it goes red the contactors open...... maybe it's the Tesla BMS opening the contactors or something but I assumed it was the Lilygo sending something to try to open them ?

dalathegreat, 2024-08-14:
@frasersimpson83 Yes, this is the battery itself taking that action. We do not know which CAN messages are needed to open the contactors.

dalathegreat, 2025-01-04:
Closing as completed I think!

closed 2025-01-04 by dalathegreat