dalathegreat/Battery-Emulator #407: Tesla Model 3 long range battery -  all components are needed in the penthouse

dvbhannes, 2024-08-12:
Do I need to have all the Penthouse original components if I want to use a Tesla Model 3 long-range battery?

Thanks
Henrik

dalathegreat, 2024-08-12:
Prefferably yes. The pyrofuse can be bypassed in a pinch.

Please read the Wiki in its entirety: https://github.com/dalathegreat/Battery-Emulator/wiki/Tesla-Model-S-3-X-Y-battery

Also there is a great example integration document at the bottom: https://github.com/dalathegreat/Battery-Emulator/wiki/Tesla-Model-S-3-X-Y-battery#example-integration

dvbhannes, 2024-08-13:
Maybe we misunderstood each other. I meant PCS – Power Conversion System (often referred to as OBC), HVC – High Voltage Controller and Low voltage connector to HVC from the vehicle (X098 connector). But I had to find out for myself that it doesn't work without the components.
So I answered the question myself! Thanks!
I'm going to do this now with a Tesla Model Y battery and a Gen24 10 Plus! I hope I succeed...

dalathegreat, 2024-08-13:
Yes, the PCS is required for smooth and easy startup. Glad it was resolved, closing this issue as completed!

closed 2024-08-13 by dalathegreat