dalathegreat/Battery-Emulator #412: Zero Motorcycle GEN2

wrightwells, 2024-08-16:
I've created a fork and added the code for the Zero 14.4 Kw battery.

Added entries in the follwing files for the ZERO_14_4_BATTERY
- -- Updated BATTRIES.h 
- -- Updated USER_SETTINGS.h 
- -- Updated webserver.cpp

I've also created a wiki in the same fork 
#https://github.com/wrightwells/Battery-Emulator/wiki/Zero-Motorcycle-14.4-Battery

There are some CAN messages still to discover and I will post a dump of the CAN messages that I have captured. 
I've still got to confirm the maeeages and get the scale factor correct for reporting but the framework is there.
However the messages so far have been taken from some other projects, 

namely: https://github.com/RIAEvangelist/zero-motorcycle-canbus

The intension would be to use the battery still within the bike rather than a standalone battery. Great for keeping at the back of the garage during the winter, as these electric bike don't really like the wet winter weather.
To do so the bike has an Anderson connector for the HV and the ability to keep the contactor open after the key is removed by adding a switch to the onboard charging connect circuit.

This is the fork location #https://github.com/wrightwells/Battery-Emulator

(It's been 25 years since I coded and it was way before GIT was invented, so I have no idea how to pull or push or commit this into the master , if and when it is ready to do so)

Also I have been whipped and shamed by the cruelest of code reviews when I posted my ccp file to ChatGPT.

nagyrobi, 2024-08-16:
@wrightwells can you also put the part numbers for the batteries known to work with your integration on the wiki page?

wrightwells, 2024-08-20:
[FAKE_BATTERY.txt](https://github.com/user-attachments/files/16676942/FAKE_BATTERY.txt)

dalathegreat, 2024-08-29:
Shall we start to open a pull request for this @wrightwells ? :)