dalathegreat/Battery-Emulator #413: Arduino IDE 2.3. edit User Settings in /* Battery settings */

dvbhannes, 2024-08-16:
I don't really understand what these parameters allow.
I have a Nissan Leaf battery with 30 kWh and a Fronius Gen24 10 plus to use
/* Battery settings */

// 300 = 30.0A , BYD CAN specific setting, Max charge in Amp (Some inverters needs to be limited)
#define BATTERY_MAX_CHARGE_AMP 300
// 300 = 30.0A , BYD CAN specific setting, Max discharge in Amp (Some inverters needs to be limited)
#define BATTERY_MAX_DISCHARGE_AMP 300

dalathegreat, 2024-08-16:
This is explained in the wiki https://github.com/dalathegreat/Battery-Emulator/wiki/Webserver-guide#battery-chargedischarge-limit

closed 2024-08-18 by dalathegreat

dvbhannes, 2024-09-26:
I still have a problem with the "contactor" so that the voltage can be measured at the HV connection so that I can be sure where plus and minus are. How do I switch the SSR 10 DD so that the Nissan Leaf battery supplies power? What else do I need to activate in the user settings options?