dalathegreat/Battery-Emulator #414: Add support for PSA/STELLANTIS e-CMP batteries (Peugeot/Opel/Jeep etc) - Help needed

albertofralbe, 2024-08-17:
Hi there,
I have a 2021 Peugeot e-208 and I would like to help the adoption of this platform's battery that is becoming increasingly common in Europe. How can I do this? 
I know that it has internal BMS using MAX17823 master and various slaves, one for each cells module.

https://openinverter.org/forum/viewtopic.php?t=2181

dalathegreat, 2024-08-18:
Hi @albertofralbe ,
See this page, we need CAN logs to get started: https://github.com/dalathegreat/Battery-Emulator/wiki/Data-needed-for-new-battery-integration

opelpanfan, 2024-08-24:
In 2 weeks I'm going get delivered one of these batteries (corsa-e 2022) also personalu own citroen e-c4 (2023) so going to be doing investigations and can logging

dalathegreat, 2024-09-28:
eCMP PR opened: https://github.com/dalathegreat/Battery-Emulator/pull/462

opelpanfan, 2024-09-28:
Varies can logs and dbc file is located on my gdrive 
https://drive.google.com/drive/folders/1S-Nf0dN5nZi71HhXIoM3GTydEk_VHHuM

Latest discovery is shunt current readings (small part is missing to discover what way current goes)

dalathegreat, 2024-12-28:
I merged PR #462 , we can continue with the Stellantis integration now in smaller bits!

closed 2024-12-28 by dalathegreat

Angel-Perez-Lopez, 2025-04-28:
Has anybody tested this integration so far?

PR #462 (2024-12-28): New battery: eCMP platform
### What
This PR adds initial, very limited, support for the eCMP Stellantis platform

### Why
Fixes #414

### How
All cellvoltages are now read and mapped properly. Candidates for SOC% and voltage also added. The code has a TODO block with things still needed before the eCMP platform can be marked Stable