dalathegreat/Battery-Emulator #415: Battery Charging Question

Protoneer, 2024-08-18:
I was just looking at how the LEAF batteries as connected to the Solar panel port on inverters (Fronius).

How does the charring of the batty work in this case? 

Also I take it the solar panels are not connected in this setup?

Protoneer, 2024-08-18:
I just noticed that the battery have their own pins...

[image]

Alizainharoon, 2024-08-18:
On this page there is an explanation of how to install and configure
https://github.com/dalathegreat/Battery-Emulator/wiki/Fronius-inverters

Protoneer, 2024-08-18:
I look on the  Fronius wiki page but it was not clear at what voltage the battery gets charged. 

Is this voltage communicated via the can bus from the battery to the inverter?

Protoneer, 2024-08-20:
Had a good dig into the code and I understand it a bit better...

Battery configuration is sent to the inverter... this includes the cell count , min/max cell voltage and from that the inverter can calculate the charging settings.

[Leaf battery set charging voltage at 400v](https://github.com/dalathegreat/Battery-Emulator/blob/ca07bb3b9d410c8a41fccce1e0de4078b2be6a8a/Software/src/charger/NISSAN-LEAF-CHARGER.cpp#L141)

closed 2024-09-28 by dalathegreat