dalathegreat/Battery-Emulator #416: Logging to SD card

wrightwells, 2024-08-19:
It would be great if you could just plug in the Liliygo to the ODB port without the use of a laptop to capture the CAN Logs as it has an onboard SD card slot.

e.g.
USER_SETTING.h
```
/* Other options */
#define LogToSD  
```

such as
https://github.com/espressif/arduino-esp32/blob/master/libraries/SD/examples/SD_Test/SD_Test.ino

Pins for the Liliygo
 * CS (SS)      = GPIO13  
 * DI (MOSI)   = GPIO15  
 * DO (MISO)  = GPIO02  
 * SCK (SCLK) = GPIO14  

you can tell my vested interest is capturing the CAN messages for the Zero motorbike rather than with a laptop in the footwell of a passenger seat. :)

**So I did it,** not elegant so will need a review but its working , no error handling on the size of the SD card

I've added it to this fork
https://github.com/wrightwells/Battery-Emulator

**---------------------------**

from arduino-esp32-master copied SD libabry etc directory to Software->src->lib

in Software->src->devboard->utils
created 
LogToSD.h
LogToSD.cpp

USER_SETTINGS.h
```
added
//#define LogToSD //Enable this line to have the TEST_FAKE_BATTERY battery used option above to write the CAN mesages to the onboard SD card.
```

Updated TEST_FAKE_BATTERY.h
added
```
#ifdef LogToSD
#include "../devboard/utils/LogToSD.h"
#include <FS.h>
char buffer[50];  
char msgString[4];
#endif
```

added to void setup_battery
```
#ifdef LogToSD
  setupLogToSD();
  writeFile(SD, "/FAKE_BATTERY.txt", "******************    Start of new messages        *************************");
#endif
```

updated receive_can_battery
```
void receive_can_battery(CAN_frame rx_frame) {
  datalayer.battery.status.CAN_battery_still_alive = CAN_STILL_ALIVE;
  // All CAN messages recieved will be logged via serial
  Serial.print(millis()); 
  Serial.print("  ");
  Serial.print(rx_frame.ID, HEX);
  Serial.print("  ");
  Serial.print(rx_frame.DLC);
  Serial.print("  ");
  
  #ifdef LogToSD
  String logData = String(millis()) + "  " + String(rx_frame.ID, HEX) + "  " + String(rx_frame.DLC) + "  ";
  #endif
  
  for (int i = 0; i < rx_frame.DLC; ++i) {
    sprintf(msgString, "%.2X", rx_frame.data.u8[i]);
    Serial.print(msgString);
    Serial.print(" ");
    
    #ifdef LogToSD
    logData += String(msgString) + " ";
    #endif
  }

  Serial.println("");
  
  Serial.println(logData); //test

  #ifdef LogToSD
  logData += "\n";
  appendFile(SD, "/FAKE_BATTERY.txt", logData.c_str());
  #endif
}
```

nagyrobi, 2024-08-19:
+1 for this!!!!

wrightwells, 2024-08-19:
didn't like it writting to the SD card constantly so I have changed the appendFile to addToBuffer .
in the LogToSD.cpp it now has a buffer and flush routine with a configurable buffer size.

needs testing before a Pull Request

wrightwells, 2024-08-20:
thats all now working but its wrapped up in the fork I created for the Zero Battery

wrightwells, 2024-08-21:
Got a wiki ready too [Write-CAN-to-SD-Card](https://github.com/wrightwells/Battery-Emulator/wiki/Write-CAN-to-SD-Card)

HVberg, 2024-11-04:
Hello
If I want to use this, do I have to copy and paste this into the Arduino IDE at USERSETTINGS.h and USERSETTINGS.cpp?and then flash the LyliGo again?
Don't want to do anything wrong. 
I am new here and trying to understand. 
Thanks

dalathegreat, 2024-12-29:
Logging to SD card now included in PR https://github.com/dalathegreat/Battery-Emulator/pull/707

closed 2024-12-29 by dalathegreat