dalathegreat/Battery-Emulator #427: SolarEdge StorEdge SE7600A-US battery emulation (LG RESU 10H)

chaseadam, 2024-08-29:
I just acquired some very low cost second hand StorEdge SE7600A-US units and am hoping I can use my existing Chevy Volt Gen1 batteries with it. These units only officially support the LG RESU 10H (HV) 400V battery (though there are "contact us" mentions for first gen powerwall).

I will keep my progress updated here, but also appreciate any guidance for extending this project to emulate the RESU 10H battery or other pointers on emulating a battery for the SE7600A inverter.

dalathegreat, 2024-08-29:
Good luck. Both battery and inverter is not supported. So there is quite the work ahead.

See the Wiki for integrating new batteries, and new inverters:
- https://github.com/dalathegreat/Battery-Emulator/wiki/Data-needed-for-new-battery-integration
- https://github.com/dalathegreat/Battery-Emulator/wiki/Data-needed-for-new-inverter-integration

ClosedLoop-NZ, 2024-09-10:
I'm in a similar situation with the AU/NZ SolarEdge inverters (SE6000-AUS20NNB2). These also support the LG RESU 10H battery. 

I'm keen to help to get this moving. I imagine what we need is a functioning LG RESU 10H system to pull CAN messages from..? The LG 10H batteries seem to be rarer than hens teeth in this corner of the world (NZ). I've been keeping an eye out over the last 6 months without any joy so may need to cast a wider net... 

Is there anyone out there with a functioning LG RESU 10H we could lean on??

dalathegreat, 2024-09-10:
We logged a SMA - RESU10H here: https://github.com/dalathegreat/Battery-Emulator/pull/398

Unfortunately, the RESU pack seems to have been configured to SMA mode, the CAN messages sent by it is appearing in the same way as other pack (BYD). So these logs wont help at all with the SolarEdge.

So, need to find a user with a SolarEdge+LG Resu10H to continue yes!

snoerenberg, 2025-05-01:
Hi,

I have a SolarEdge SE5000H (AC Inverter) with SE
Sesti-S4 to charge 2x LG 10 H Prime.

snoerenberg, 2025-05-01:
The batteries communicate via RS-485 with the Sesti-S4.

chaseadam, 2025-05-02:
> The batteries communicate via RS-485 with the Sesti-S4.

@snoerenberg any chance you can capture the RS-485 per https://github.com/dalathegreat/Battery-Emulator/wiki/Data-needed-for-new-inverter-integration

FilippoChiappo, 2025-05-07:
Hi, you can try to decode the messages using this [Modbus Definition List](https://photomate.zendesk.com/hc/en-gb/articles/4573250566941-Solar-Inverter-Modbus-Definition-List), it contains also commands for LG Resu batteries.

snoerenberg, 2025-05-08:
Hi @chaseadam, @FilippoChiappo I used a Raspberry PI 3b+ and an USB RS485 Modbus Converter.

Similar to this one: 
[image]

I used grok to create a sniffer and try to parse the captured data as modbus.
This seems to be partially fine.

It could get 3 devices with ID: 2 (Inverter), 14 (Battery 1), 15 (Battery 2).

Plain log parsed upon modbus specifics:
[rs485_log.txt](https://github.com/user-attachments/files/20102608/rs485_log.txt)

Analysis of the captured log:
[rs485_analysis.csv](https://github.com/user-attachments/files/20102609/rs485_analysis.csv)

Scripts used:
[python_scripts.zip](https://github.com/user-attachments/files/20102618/python_scripts.zip)

What I can do? Anyone with experience in this?
Currently it should just contain charging one battery.

FilippoChiappo, 2025-05-08:
Wow, good start!
Please, record the communication when certain events occurs, like is written here [link](https://github.com/dalathegreat/Battery-Emulator/wiki/Data-needed-for-new-inverter-integration#i-have-the-logger-and-a-working-system-whats-next)

To reduce the complexity for the deconding phase, is useful to get logs only when one battery is connected. Please, disconnect the RS485 cable from the second battery.

Another check that you can try to do is:
When the system is working (only one battery) for example in charging, try to disconnect the communication cable from the battery and see what happens. This is useful to understand if the rs485 is necessary to regulate the charging / discharging or only to check the info (SoC, temp).

Because, I think that the control of the power flow is done "natually" by the bidirectional DC/DC Converter, after chosing the DC bus voltage

snoerenberg, 2025-05-13:
I switched to the following sniffer: https://github.com/alerighi/modbus-sniffer

This provides an output possibility as Wireshark pcap file which improves the analysis and filtering capability.

e.g. filter for Unit and frame length:
[image]

You just need to configure the DLT_USER protocol settings:
[image]

Captures:
[SolarEdgeLG.zip](https://github.com/user-attachments/files/20187833/SolarEdgeLG.zip)

still 2 batteries included because I cannot simply disconnect it, I might need the electrician to configure it properly with one battery.

The 65 Byte Frame contains most likely:
Register
-> 201 -> 1=standby, 3=active
-> 203 -> charge/discharge watt (Signed Int)
-> 221 -> SoC

snoerenberg, 2025-12-26:
@bob-atkins I stopped with the above initial analysis and supplied the capture of it.

I have two LG batteries attached to one SolarEdge inverter, this creates a bit of overhead in the communication.

If it is urgently necessary I could unplug on of them and recreate a capture. With current weather conditions in Germany, it will be not easy to capture a fully loaded battery. Or some cycles.

Maybe you can start with the capture I supplied.

Stephan