dalathegreat/Battery-Emulator #429: Tesla3 battery without PCS and discharge isues.

StigWi, 2024-08-30:
Hi,
I have a Tesla battery and a Fronius gen24 plus 6 inverter.

The PCS i burned, but BMS is sending signals, so I connected the battery to the inverter without using the contactors,
My inverter has charged the battery to 85% but is not discharging the battery.
The battery emulator says max discharges power 0W.
Can I change it anywhere or is it a restriction from the BMS, that not accept the inverter to discharge the battery?

I have attached 2 pictures.

Thanks in advance and for a very interesting projekt.
Stig

[image]
[image]

StigWi, 2024-08-30:
Hi Dalathegreat
You have solved my problem for the last 3 weeks.
Now I can sleep again.
Thanks a lot:)
Regards
Stig

dalathegreat, 2024-08-31:
Nice! I will close this issue as completed, and make a note in the Wiki about this (needed incase the PCS is not used)

closed 2024-08-31 by dalathegreat