dalathegreat/Battery-Emulator #430: Tesla model 3 Troubleshooting Hvil & Current

FreReg, 2024-08-31:
Hello! 
I have a problem with my installation, I have a Tesla model 3 battery 2022 RWD - China manufactured pack - LFP cells - Part number 1666969-00-D and a Goodwe GW25K-ET

I feel like I have several strange problems!

1. Rest Not ok: 
the loop between terminal 7.24 is closed and complete to PCBA "BMS" & terminal 10.16 as well.

2. pyro fuse: replaced and has approx. 2 ohms up to the terminal on the PCBA.

3. CAN Bus: gets out info about battery volts etc. if I measure the Can bus resistance I get it to 1.54MΩ feels wrong should be 120Ω? when battery and inverter are plugged in at the same time, the battery interferes with Can bus communication in the inverter in the goodwe app, the battery disappears completely

4. If I check liligon's web interface, it shows -3276.8A & Power 1114.1kW but no load connected. the shunt replaced with a new one but no difference.

60Ω connected across terminal 1.3 on X098 = RWD

Condenser connected according to instructions.

can anyone give me some tips or a solution mine is running out now...

[image]
[image]
[image]

dalathegreat, 2024-08-31:
Hej FreReg!

I am no Tesla expert, so others might be better to chip in. But atleast to answer your question 3., the Tesla pack does not have a terminating resistor, so you have to add 120 Ohm over CAN-H and CAN-L pins on the battery side to get the termination correct. The sum of termination resistors should be 60Ohm, one in each end of the bus.

(It feels like we have a biweekly HVIL troubleshooting on the [Discord server ](https://www.patreon.com/dala):sweat_smile: )

FreReg, 2024-08-31:
hello Dala! 
A resistor is mounted but there is no difference , have read on a page that the current must be activated in the can bus commands do you know if it is included in the code have not yet checked it?

dalathegreat, 2024-08-31:
The current being -32768 (essentially value sent as 0xFFFF) usually indicates a broken sensor or damaged wiring. I have never seen this on a Tesla battery before.

To me, apart from the current sensor issue, it looks like it fails the HVIL check (HVIL: NOT OK), and thus skips to proceed further.

josiahhiggs, 2024-09-01:
Hi FreGeg, I have been helping a few through the HVIL trouble shooting on the Discord, which is absolutely the most awesome group of people! 
The first thing to get fixed is the HVIL circuit. One circuit of the HVIL circuit is the grounding of the two fasteners at the X098, This is checked to confirm the penthouse is secured. So make sure the penthouse lid is bolted to the battery and at the X098.
The Tesla BMS does not have a terminating resistor in the CAN H - L circuit and will communicate directly to the Lilygo with the 120 ohm resistor on the board. The inverter should have a 120 ohm resistor termination. So Lilygo + Inverter = 60 ohms.
I hope you have the PE connected between the inverter and battery and that the CAN H - L are connected correctly and not reversed on the inverter.

FreReg, 2024-09-02:
Hi josiahhiggs! 
I tried everything, see picture below. 
I have measured the cable that goes to the shunt, it is also intact and replaced the shunt with a new one, no difference. I think something is wrong in the HV controller, like that the voltage up to the shunt and HVIL is not activated there because the HV of the HVIL loop is not detected, and I get sensor error on the shunt!
[image]

dalathegreat, 2024-09-06:
@FreReg It is rare, but one person had a bricked PCS, after replacing it inside the penthouse, the battery sprung alive...

doliban, 2024-09-13:
Hi
Have the same problem
Checked everything, tried different resistors, replaced the BMS, replaced the PCS.  
But HVIL not OK, the contactors do not close.

dalathegreat, 2024-10-30:
@doliban , Model X requires Digital HVIL. We recently added this: https://github.com/dalathegreat/Battery-Emulator/pull/551

dalathegreat, 2025-05-05:
Closing old issue, SW has improved quite a bit for troubleshooting since 7.0.0! Make new issue if needed!

closed 2025-05-05 by dalathegreat