dalathegreat/Battery-Emulator #441: Richiesta integrazione batteria Dacia Spring

vidmack, 2024-09-06:
Buongiorno chiedo se è possibile di integrare anche la batteria della Dutchess Spring che ho un pacco Batterie che vorrei tanto collegare l’inverte e ho comprato gel Lily Go e con le batterie esistente non riesco a comunicare mi danno dei caratteri strani ho provato anche Renault Zoe che dovrebbe essere molto simile ma purtroppo non riesco a comunicare Se qualcuno ha già provato o questa esperienza quando dacia Spring per favore di scrivere O se c’è bisogno di qualche dato da recuperare dalla macchina o la seconda macchina che è funzionante dacia e se può se posso scrivere dei comandi e Cannes tramite Lil Go sono disposto di farlo soltanto ad avere un piccolo istruzione vi ringrazio in anticipo.

nagyrobi, 2024-09-06:
Perhaps in English?

dalathegreat, 2025-03-15:
Dacia Spring integration: https://github.com/dalathegreat/Battery-Emulator/pull/971

Wiki: https://github.com/dalathegreat/Battery-Emulator/wiki/Battery:-Dacia-Spring-%E2%80%90-Renault-K%E2%80%90ZE

vidmack, 2025-03-19:
Hi, thanks for the work you are doing, I wanted to ask you if you can also
insert a new Solxpow x3 inverter, the parent company sent me the Canbus
communication protocol, see if you can do something, thanks

dalathegreat, 2025-03-19:
@vidmack 
Please make a new issue requesting this integration. Attach the files there also, it will speed up the integration.

PR #971 (2025-04-13): New battery: Add support for CMFA-EV battery
### What
This PR implements support for the CMFA-EV platform

### Why
To add support for [Dacia Spring / Renault K‐ZE batteries ](https://github.com/dalathegreat/Battery-Emulator/wiki/Battery:-Dacia-Spring-%E2%80%90-Renault-K%E2%80%90ZE). User requested feature, fixes #441 and fixes #338

### How
Info based on the log files provided by Fbostrom on the Discord server :raised_hands:
closed 2025-04-13 by dalathegreat