dalathegreat/Battery-Emulator #442: Polestar2 CAN battery contactors not working

DonatelloXX, 2024-09-07:
We have wired up a Polestar2-battery with a Lilygo-Board and a Fronius Symo Gen24 Plus.

Communication over Modbus and CAN-Bus is ok. The Webserver is showing everything green and "Automatic contactor closing allowed" is checked for Battery and inverter.

But the contactors of the battery won't open. there is no voltage from the battery.

Are we doing something wrong or is this simply not included in the software? Wehen not included - what should we provide to get support for this?

dalathegreat, 2024-09-07:
I checked, and the whole Volvo support was developed by one user over on the [Patreon Discord](https://www.patreon.com/dala) server. The battery was already disassembled, so contactors were skipped and external ones used
[image]

So Volvo SPA support is kind of still experimental , since there is only 1 user with a Volvo/Polester battery setup. But all the values work, but the Contactor section is not working yet.

To get the contactors engaged, we will need a vehicle CAN log from a functional vehicle, which we can then replay towards the battery.  See the [CAN logs needed](https://github.com/dalathegreat/Battery-Emulator/wiki/Data-needed-for-new-battery-integration#can-logs-needed) section of the wiki.

Alternatively, bypassing the contactors or [driving them via the GPIO pins](https://github.com/dalathegreat/Battery-Emulator/wiki/Contactor-Control-via-GPIO-pins) can be done.

DonatelloXX, 2024-09-08:
Thank you for the quick response!
This makes absolute sense now 😉
We will try to get CAN-logs from an intact Polestar 2 to use the internal contactors.

DonatelloXX, 2024-09-09:
We were able to Log the CAN-Output with the Lilygo via Putty to a File. (Log all Session Data)

I attach 2 Logfiles which are still in binary Format. I know too little about the CAN-Protocol and hope that someone can extract the data needed to control the battery contactors.

This are the actions taken between start and stop logging:

- unlock vehicle with key
- Brake
- wait for ready state
- engage reverse gear
- drive 30cm backwards
- brake
- engage forwared
- drive 30cm forward
- brake
- parking mode
- open door
- switch off car with Infotainment
- close door
- lock vehicle with key
- Wait for opening contactors
- Waiting for bus silence

[Polestar Log3.txt](https://github.com/user-attachments/files/16933689/Polestar.Log3.txt)
[Polestar Log4.txt](https://github.com/user-attachments/files/16933699/Polestar.Log4.txt)

dalathegreat, 2024-09-10:
Hi @DonatelloXX 
Unfortunately that log file made by ChatGPT is complete hallucination as far as I can tell.

I made a quide on the Wiki on how to log with the LilyGo: https://github.com/dalathegreat/Battery-Emulator/wiki/CAN-logging

DonatelloXX, 2024-09-10:
@dalathegreat The Hint about the boud-rate would be useful on https://github.com/dalathegreat/Battery-Emulator/wiki/Data-needed-for-new-battery-integration#taking-can-logs-with-the-battery-emulator-lilygo-hardware too 🥺

I will get new logs ASAP 😉

DonatelloXX, 2024-09-12:
Here are some new logs. Taken with 115200 Boud.

- Full cycle until bus silence [Polestar.txt](https://github.com/user-attachments/files/16982630/Polestar.txt)
- Wakeup from bus silence [Polestar wakeup.txt](https://github.com/user-attachments/files/16982629/Polestar.wakeup.txt)
- Start and stop charging via AC [Polestar charge.txt](https://github.com/user-attachments/files/16982631/Polestar.charge.txt)

Hope these logs are more useful 😊

DonatelloXX, 2024-11-24:
We managed to switch the contactors manually. It was necessary to open the Relay-Box and connect 12V to the contactors. Be aware to disconnect the contactor on the right hand side of the battery to ensure there is no voltage on the relay-box.

DonatelloXX, 2024-11-24:
Mark as resolved

closed 2024-11-24 by DonatelloXX

GoSmartEV, 2024-12-29:
> We managed to switch the contactors manually. It was necessary to open the Relay-Box and connect 12V to the contactors. Be aware to disconnect the contactor on the right hand side of the battery to ensure there is no voltage on the relay-box.

there is good news, the contactors were able to engage, so soon there will be an update where the battery will be added as fully operational through CAN ✅

dalathegreat, 2025-01-03:
Contactor closing now implemented in PR https://github.com/dalathegreat/Battery-Emulator/pull/744