dalathegreat/Battery-Emulator #443: PYLON CAN address

gkasprow, 2024-09-07:
How do we change the Pylon battery address? I can't find it in files.
I need to add a second battery to SOFAR - my 20kW inverter has two battery channels.
But I need to change the CAN address.
I assume the battery stays connected to the native ESP port, and the inverter is connected to the external CAN controller?

dalathegreat, 2024-09-07:
The defines SEND_0 and SEND_1 can be toggled in PYLON-CAN.cpp https://github.com/dalathegreat/Battery-Emulator/blob/3d090ca7bffa026283c9cf72be7d366e947e82b9/Software/src/inverter/PYLON-CAN.cpp#L6

From the Sofar wiki: https://github.com/dalathegreat/Battery-Emulator/wiki/Sofar-inverters#inverter-configuration
_In "Advanced Settings/Battery Parameters" choose Pylontech as battery type and set 01 as battery address (00 is default), save your settings. Check all fuses and everything else twice, then turn on Lilygo, and voila!_

I would assume that to run two packs, you need to run one as ID 00, and the other one as ID 01. Not sure how Sofar works, you might need to have 2x separate LilyGo systems.

gkasprow, 2024-09-07:
Thanks! That was not obvious :) Yes I will need two Liligo systems.

gkasprow, 2024-09-07:
Another question. The battery should be connected to external CAN controller and the inverter must be connected to embedded CAN controller? Or, that can be defined by software?

dalathegreat, 2024-09-08:
That can be defined in the software, in USER_SETTINGS
[image]

closed 2024-10-30 by dalathegreat

gkasprow, 2025-05-01:
I've just discovered that I'm using version 5.9 :D A lot changed since that time..