dalathegreat/Battery-Emulator #444: HA, MQTT: login and PW

shatreyman, 2024-09-08:
Hi!
New user here. All works, but would like to get data correctly to HomeAssistance. Assume mqtt is a good way. How to I setup MQTT login and pasoword for lilygo device? I see it is trying to send data, but in LOG such text:

```
2024-09-08 21:12:33: New connection from 192.168.XX.XX:61607 on port XXXX.
2024-09-08 21:12:33: Client LilyGoClient-esp32-67DEXX disconnected, not authorised.
```

cant find any place in configuration file where PW and login can be setted up.

dalathegreat, 2024-09-08:
Hi @shatreyman , this is done in the USER_SETTINGS.cpp file

[image]

Here you can specify the username and password.

I see this info was missing from the [MQTT wiki](https://github.com/dalathegreat/Battery-Emulator/wiki/MQTT), I will add it there also!

shatreyman, 2024-09-09:
Literally looked everywhere except this file. Thanks! Works well now

closed 2024-09-09 by shatreyman