dalathegreat/Battery-Emulator #450: Tesla Battery 1521262-00-A support

dvbhannes, 2024-09-09:
Hello, would like to buy a Tesla battery 78 kwh with the number 1521262-00-A. They can't be bought quite cheaply. Does this work with the Battery Emulator????

dalathegreat, 2024-09-09:
Seems like nobody has tested that battery yet. As long as it is 2020+, it should work well with the emulator. See the Wiki for more info: https://github.com/dalathegreat/Battery-Emulator/wiki/Tesla-Model-S-3-X-Y-battery#which-packs-work

josiahhiggs, 2024-09-17:
I believe you are referring to the Model 3 Long Range Dual Motor 78kWh NCM Battery. This should be supported as others have successfully used the 75kWh and 78kWh long range versions. Post a picture of the battery you are interested in buying and we could compare to other known working part numbers.

Floriankon, 2024-10-04:
Guten Tag, würde diese Tesla Batterie funktionieren? 1104423-00-M
[image]

Floriankon, 2024-10-04:
[image]

dalathegreat, 2025-01-15:
Yes.
Closing this as completed

closed 2025-01-15 by dalathegreat