dalathegreat/Battery-Emulator #475: Lead acid support for DEYE inverter?

mace-de, 2024-09-21:
Would it be possible to add a Support for lead acid batteries on Deye Inverters?
The two main parameters battery voltage and current could be read from the inverter Modbus registers 587 and 591. This makes it obsolete to acquire them by hardware. Maybe a NTC or DS18B20 temperature sensor could be used on a port pin for temperature compensation.
All other Parameters like
- Capacity
- Absorption Voltage
- Float Voltage
- Equalisation Voltage
- Temperature coefficient
- Battery resistance
- Charging efficiency
- Maximum allowable discharge level
are constant has to be configured by the user. 
With these parameters a charging curve and a SOC could be calculated to be transferred as BMS-Data.
I think I am able to stumble the above together as a separate arduino project with no additional features and unusable for others. But maybe it's better to have this included here. I for my part, I'm definite not able to integrate this in your project code.

bhunziker, 2024-10-02:
Hi Oliver
I believe this is a great idea. By integrating your approach in this great project it would become very versatile. Not only any kind of battery could be used but also the already integrated inverters. I'm not a software expert but would like to support with testing and documenting.
Regards Bernhard

mace-de, 2024-10-02:
I did some test the last days. I have the Modbus running and do some calculations for remaining AmpHours and SOC for my battery. The results are shown on a small display. This seems to work so far. But this is a separate project and not included in the Battery-Emulator source until now.

mace-de, 2024-10-30:
Have the first version finished and running. It's a bit hacked together and worse coded but it's working as intended. Have uploaded a repo if someone is interested.

casper2222, 2025-01-14:
> Have the first version finished and running. It's a bit hacked together and worse coded but it's working as intended. Have uploaded a repo if someone is interested.

Hi , I am interested, where can i download?

casper2222, 2025-01-15:
> Because github is forcing to use 2FA soon, it's hosted at codeberg.org now. https://codeberg.org/mace-de/Deye-HV-Lead-Acid-BMS-Emu

Thanks, can you write to me at casper2@bahnhof.se
Would have to ask a few things if you have time?

dalathegreat, 2025-02-16:
Closing this one as "wont do" for the Battery-Emulator, as a workaround seems to exist in https://github.com/dalathegreat/Battery-Emulator/issues/475#issuecomment-2591094100

closed 2025-02-16 by dalathegreat