dalathegreat/Battery-Emulator #488: Configuration User Settings for Nissan Leaf and Gen24

dvbhannes, 2024-09-27:
I have a Nissan Leaf battery with 30 kWh and a Gen24 10 Plus. I flashed the Lilligo for the emulator and in the user settings I configured the
1.
#define NISSAN_LEAF_BATTERY
2.
#define BYD_MODBUS
3.
#define HW_LILYGO
4.
#define DEBUG_VIA_USB
5.
#define DEBUG_CANFD_DATA
6.
#define CONTACTOR_CONTROL
.
Lilligo is green.
The Gen24 10 Plus switches constantly and the connector does not switch via the semiconductor relay, so I cannot measure any current at the high-voltage connection on the Leaf battery.
What am I doing wrong???

- Software version: 7.0.0
- Battery used: Nissan Leaf 30 kwh
- Inverter communication protocol: Modbus / Gen24
- Hardware used for Battery-Emulator: HW_LILYGO,
- CONTACTOR_CONTROL: `yes/no` ????
- DUAL_CAN: no
- WEBSERVER: yes - green
- MQTT: no

dalathegreat, 2024-09-27:
These two options are not needed: DEBUG_VIA_USB & DEBUG_CANFD_DATA

Check the Webserver. Do you have a green checkbox on the contactor closing allowed? https://github.com/dalathegreat/Battery-Emulator/wiki/Contactor-Control-via-GPIO-pins#troubleshooting-

Some LEAF batteries dont always set the check ON. If you have a :x: on the Battery side, then follow the instructions: on the LEAF page to force datalayer.system.status.battery_allows_contactor_closing = true; https://github.com/dalathegreat/Battery-Emulator/wiki/Nissan-LEAF-battery#automatic-control-

dalathegreat, 2024-09-27:
@dvbhannes If you have two checkboxes for both Battery and Inverter, the contactor control via GPIO will spring to life!

The polarity is shown in the Wiki
[image]

dvbhannes, 2024-09-29:
Thanks! I have now installed 7.3.0.
I have now entered datalayer.system.status.battery_allows_contactor_closing = true in line 745 of the file Software/src/battery/NISSAN-LEAF-BATTERY.cpp. Now the first relay (LED) flashes briefly and then relays 2 and 3 light up continuously and the voltage at the HV output can be measured. The Gen24 starts and wants to charge the Nissan Leaf battery, but then stops and the green LED on the Gen 24 turns bright green. You can hear the relays clicking...
Here are my screenshots from the web server.
[image]
[image]

What could be the reason that the Gen24 keeps crashing and restarting...?

dalathegreat, 2024-09-29:
Nice work! Next up is troubleshooting on the Gen24 side. Post any fault codes you see, and inverter status. Here are some starting tips for troubleshooting: https://github.com/dalathegreat/Battery-Emulator/wiki/Fronius-inverters#troubleshooting

dvbhannes, 2024-09-29:
Unfortunately, I noticed that there is no voltage at the HV connection. I measured the negative pole against the chassis ground with the multimeter. The multimeter showed me voltage. The voltage measured at the HV connection was probably just potential...?? The problem seems to be the Nissan Leaf battery. When the SSR relays switch, should you hear a switching noise from the battery??? Unfortunately, I don't hear any clicking...!

dalathegreat, 2024-09-29:
There should be a faint click. Make sure you are on the right pins (RLY N and RLY P), you can also remove the SSRs ad manually apply 12V to the pins on the connector to see if the relays click. When you have confirmed by manually applying 12V between RLY P V - RLY P GND to hear the click, see if the SSR does it too.

Also make sure the service disconnect switch is seated on the battery.

dvbhannes, 2024-09-30:
The service circuit breaker is installed. It may be that the Nissan Leaf battery relay only works in a horizontal position. I mounted the Nissan Leaf battery vertically on the wall...
[image]

dalathegreat, 2024-09-30:
No that is not a problem, we have over 10 people with pictures on horizontally mounted LEAF packs on the [Patreon Discord server](https://www.patreon.com/dala)

I suggest manually supplying 12V to the following pins: 

[image]

dvbhannes, 2024-10-04:
Thanks for the help. I found the error. The SSR-10 DD only switches at 3.85 volts. So I have to try other SSRs. The relays from the Nissan Lea battery switch properly if I manually bridge the positive! The LED of the SSR-10 DD lights up at 3.22 volts but only switches at 3.85 volts. Not good quality from China!!
[image]

dvbhannes, 2024-10-16:
Hello Dala, I have installed the new SSR from Fotek. Now the relays on the Nissan Leaf battery have switched correctly and everything is working! Thank you very much for your good work! I really appreciate your work!!!! The setup with the Nissan Leaf battery has been working perfectly for a week. Now I will connect my Tesla 78 kWh battery and hope that it works well too!!! Thanks!!!
[image]

closed 2024-10-16 by dvbhannes

ougueur, 2024-12-18:
Bonjour .
A quelle puissance charge ta batterie en charge forcée ? 
J arrive a chargé a puissance maximum  avec le surplus des panneaux , mais en charge forcée ( mode manuel - charge minimum a 100%- autorisation de charge sur réseau public activé)  la puissance charge ce limite environ 500w 
Gen24plus primo 6k 
Batterie leaf30  protocol byd 11hvm modbus 

[image]
[image]

dalathegreat, 2024-12-19:
Hi, you need to use the "Time-dependent battery control" incase you want to charge the battery from grid:

[image]