dalathegreat/Battery-Emulator #492: OTA updates causes RESET_PANIC

StarkJohan, 2024-09-28:
When updating OTA the percentage halts around 10% and causes a RESET_PANIC. This is consistent on release 7.3.0 and above but works as intended on 7.1.1. When uploading by wire everything works as intended.

Using Stark CMR v1 as it's a two can-bus setup so I can't test the exact settings on the lilygo.

volatile CAN_Configuration can_config = {
    .battery = CAN_NATIVE,   // Which CAN is your battery connected to?
    .inverter = CANFD_NATIVE  // Which CAN is your inverter connected to? (No need to configure incase you use RS485)
    //.battery_double = CAN_ADDON_MCP2515,  // (OPTIONAL) Which CAN is your second battery connected to?
    //.charger = CAN_NATIVE                 // (OPTIONAL) Which CAN is your charger connected to?
};

#define NISSAN_LEAF_BATTERY
#define BYD_CAN 
#define HW_STARK
#define CONTACTOR_CONTROL  
#define CAN_FD
#define USE_CANFD_INTERFACE_AS_CLASSIC_CAN

#define WIFI
#define WEBSERVER
#define WEBSERVER_AUTH_REQUIRED false 

#define LOAD_SAVED_SETTINGS_ON_BOOT 
#define HA_AUTODISCOVERY (to avoid the ifdef bug in this version, not using MQTT)
#define DUMMY_EVENT_ENABLED false

#define BATTERY_WH_MAX 58000
#define BATTERY_USE_SCALED_SOC true
#define BATTERY_MAXPERCENTAGE 8000
#define BATTERY_MINPERCENTAGE 2000
#define BATTERY_MAX_CHARGE_AMP 750
#define BATTERY_MAX_DISCHARGE_AMP 750

PaulV71, 2024-09-28:
We have the same, using 7.3.0
[image]

StarkJohan, 2024-09-28:
> We have the same, using 7.3.0 ![OTA reset_panic event](https://private-user-images.githubusercontent.com/181633608/371756772-70c6cc58-2363-4a5d-b831-a13dc09d5a76.JPG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjc1Mjc1NjQsIm5iZiI6MTcyNzUyNzI2NCwicGF0aCI6Ii8xODE2MzM2MDgvMzcxNzU2NzcyLTcwYzZjYzU4LTIzNjMtNGE1ZC1iODMxLWExM2RjMDlkNWE3Ni5KUEc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTI4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkyOFQxMjQxMDRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yYjliNTRmY2EyYWYxNmRiOTkzYTkyMDdjZmE0NzA4NWNhMmNhZmY2NThhODIxMTJhMTMzNDUwMTU5NmIzNjNkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.73cSo6DPgXNtM24qC3ld1cwCEpzBPos1DajQniR8VqQ)

What hardware and software version? What setup?

PaulV71, 2024-09-28:
Lilygo Tcan connected to the LEAF battery. The other one to the inverter has no issues so far.

dalathegreat, 2024-09-28:
@amarofarinha could this be related to the OTA library update? https://github.com/dalathegreat/Battery-Emulator/pull/454

StarkJohan, 2024-09-29:
@amarofarinha solved this and I have confirmed that his solution works.

PaulV71, 2024-09-29:
problem solved. Thanks