dalathegreat/Battery-Emulator #500: Liligo freezes - solid green light

bubble07, 2024-09-29:
### Expected Behavior
Stable performance, or enhanced crash diagnostics

### Actual Behavior
Issue:
3 instances of Liligo freezing within an hour 

Symptoms:
Inverter reports no BMS comms
Liligo webserver not available
MQTT signals stop
Status LED on Liligo is SOLID green
Inverter can still see HV battery voltage, so I believe the contactors are remaining closed

Upon turning power off and on (to Liligo and battery), everything starts up as normal with no events stored other than normal start up.
System is Kia - FoxESS with Lili and MCP add on CAN.
Running V7.3 dev when it first crashed, then recompiled with latest main (V7.4 dev) but it crashed again twice.

After third crash, system has been OK for 5h so far.

No serial logs available, but MQTT sensors have been recorded by home assistant and are available if helpful, let me know what would help.

### Steps to Reproduce the Problem
Unable to reliably provoke the issue. Next steps of diagnosis will be to try using same software and settings with a different Liligo and MCP chip.

First crash occurred shortly after the end of an overnight charge cycle to 95% scaled SoC (approximately 77% real)
[SoC During Crash Events.pdf](https://github.com/user-attachments/files/17179005/SoC.During.Crash.Events.pdf)

### Settings

User setting files attached
[USER_SETTINGS_cpp.txt](https://github.com/user-attachments/files/17179002/USER_SETTINGS_cpp.txt)
[USER_SETTINGS_h.txt](https://github.com/user-attachments/files/17179003/USER_SETTINGS_h.txt)

- Software version: `7.3-dev` & '7.4-dev'
- Battery used: `Kia`
- Inverter communication protocol: `SOLAX`
- Hardware used for Battery-Emulator: `HW_LILYGO`
- CONTACTOR_CONTROL: `yes`
- DUAL_CAN: `yes`
- WEBSERVER: `yes`
- MQTT: `yes`

dalathegreat, 2024-10-04:
If I had to guess what is happening here, I would say something related to Solax interfacing with the MCP2515 chip, causing a crash.

I have a few suggestions you could do:
- Switch around where battery and Solax is connected.
- Install an CAN bus isolator on the Solax connected channel, more info here: https://github.com/dalathegreat/Battery-Emulator/wiki/Lightning-strike#suggested-hardware

bubble07, 2024-10-04:
Thanks, for the guidance, I will get ordering.
When you say switch around, you mean have the Lili interfering with the inverter and the MCP2515 interfering with the battery?

dalathegreat, 2024-10-05:
@bubble07 Yes just that!

bubble07, 2024-10-05:
Thanks, I will try that later. I have just got back to the battery and tried the second set of hardware I have:

Test 1: Lili#2 (battery) with MCP2515 #2 (inverter):
Lili sees battery ok, contactors close however inverter doesn't see battery (no messages get through), possibly a faulty MCP2515.

Test 2: Lili#2 (battery) with MCP2515 #1 (inverter):
Same as original condition (lili#1 & mcp2515#1), system functions as expected for a short period of time (5-10 minutes) before Lili crashes with solid green light.

So the issue is reproducible on different liligos boards, this backs up your suggestion that the issue is MCP2515 or inverter related.

Next test will be as you suggest, swapping lili and MCP2515 roles. I will also try to prepare another MCP2515 with the correct can chip to rule in/out a fault with MCP2515#1.

bubble07, 2024-10-06:
I tried with 7.5 dev with the roles reversed (HV DC isolated):

First I left the inverter CAN disconnected, to see if it would reliably run, lili could see battery SoC and was shown on web server and MQTT, but after about 1.5 minutes LED went red with the following events:
CELL_DEVIATION_HIGH
WATER_INGRESS

Then I rebooted with inverter connected, all seemed good to start with: green LED, no error events, battery SoC shown at the inverter, then after approx 2.5 minutes LED goes yellow and a lot more unexpected error events, none of which have been seen before with the Lili attached to the battery:
12V_LOW
WARNING
06/10/2024, 08:42:05
1
12
12V battery source below required voltage to safely close contactors. Inspect the supply/battery!
BATTERY_OVERHEAT
ERROR
06/10/2024, 08:42:05
1
116
ERROR: Battery overheated. Shutting down to prevent thermal runaway!
BATTERY_EMPTY
INFO
06/10/2024, 08:42:02
1
0
Info: Battery is completely discharged
BATTERY_UNDERVOLTAGE
WARNING
06/10/2024, 08:41:48
1
193
Warning: Battery under minimum design voltage. Charge battery to prevent damage!
BATTERY_FULL
INFO
06/10/2024, 08:41:28
2
0
Info: Battery is fully charged
SOH_LOW
ERROR
06/10/2024, 08:41:08
1
0
ERROR: State of health critically low. Battery internal resistance too high to continue. Recycle battery.

@dalathegreat Could this be a faulty MCP2515 chip?

bubble07, 2024-10-06:
I replaced the MCP2515 and it appears to be working as expected. System has been running for 4.5h so far without a crash or unexpected events.
Thanks for the support and suggestions Dala.👍

closed 2024-10-06 by bubble07