dalathegreat/Battery-Emulator #501: Emulator not responding to can messages for PYLON_CAN and CANFD setup.

StarkJohan, 2024-09-29:
### Expected Behavior

The emulator should respond with "send_setup_info()" and "send_system_data()" when receiving the pylon frames.

### Actual Behavior

According to the canfd debug output the emulator is not responding at all to the frames as can be seen in the attached image.

### Settings   

Latest main as of today 2024-09-29 7.4.dev (using the RX+TX canfd-debug addon merged into main)
.battery = CAN_NATIVE,   
.inverter = CANFD_NATIVE, 
#define BYD_ATTO_3_BATTERY
#define PYLON_CAN 
#define HW_STARK
#define DEBUG_CANFD_DATA 
#define CONTACTOR_CONTROL
#define CAN_FD 
#define USE_CANFD_INTERFACE_AS_CLASSIC_CAN 
#define WEBSERVER
// #define MQTT 

Battery on can1, inverter on can2 (canfd native)
[image]

dalathegreat, 2024-10-04:
I tried to replicate this with my Stark board, on bootup I get

CAN FD add-on (ESP32+MCP2517) selected
CAN-FD Configuration error 0x1
CAN-FD initialization failed. Check hardware or bitrate setting

+ The event

[image]

StarkJohan, 2024-10-04:
This is a report from another Stark board user. I haven't actually tested this on my own (yet) but this should mean that we have some kind of conflicting settings in this combination of inverter and battery, right?
On the other hand, if initialization fails, the emulator should not see any data on the bus at all?

dalathegreat, 2024-10-04:
Hmm, I def had the CANFD port working on an old dev version. Something might have broken it.

When looking at the COnfiguration Error 0x1, it states
_19.3.1 kRequestedConfigurationModeTimeOut
The ACAN2517FD::begin method first configures SPI with a 1 Mbit/s clock, and then requests the configu-
ration mode. This error is raised when the LCP2517FD does not reach the configuration mode with 2ms. It
means that the MCP2517FD cannot be accessed via SPI._

Yeah this error seems different from the user above, since they can print USB messages...

StarkJohan, 2024-10-04:
I have 7.4.dev from earlier this week running just fine with can+canfd with BYD_CAN + LEAF so it has to be config related.

dalathegreat, 2024-10-04:
Found the bug, it is related to CANFD_NATIVE / CAN_ADDON_FD_MCP2518

Configuring the Stark board to CAN_ADDON_FD_MCP2518 makes everything work

dalathegreat, 2024-10-05:
Closing as completed, fix in main. Will be included in next release (v7.4.1 / v7.5.0)

closed 2024-10-05 by dalathegreat

StarkJohan, 2024-10-05:
Confirmed working on the affected system!