dalathegreat/Battery-Emulator #502: Onduleur Deye compatible EU 6KW single phase

alstare72, 2024-09-29:
Bonjour a tous .. 

J'ai DEYE SUN-6K-SG03LP1-EU monophasé 6kw .. j'ai déjà 2x280ah (30kw) box basengreen j'ai besoin d'augmenter .. 

Est ce que vous pensez que je peux utiliser le travail que vous avez effectué sur les autres modèle de deye .. 
Je cherche depuis longtemps comment utiliser les batteries de EV .. 

MERCI d'avance 
To France

mvgalen, 2025-01-02:
Google translate result:
I have DEYE SUN-6K-SG03LP1-EU single phase 6kw.. I already have 2x280ah (30kw) basengreen box I need to increase..

Do you think I can use the work you did on the other deye models..
I have been looking for a long time how to use EV batteries..

mvgalen, 2025-01-03:
The inverter you mention is only suitable for low voltage batteries (40-60v), so to be able to use a EV battery, you will need another inverter.

closed 2025-01-03 by mvgalen