dalathegreat/Battery-Emulator #527: how to begin to make a battery Emulator for Opel Ampera E / Chevrolet Bolt 64 Kw

HVberg, 2024-10-10:
Hi Im new here and like to learn but dont know how to start this project
First I wanted to learn how i can make the needed logfiles from the working car and the stand alone battery, I have those both.
i read almost everything but dont know yet how i can make the logfiles with the LiLygo?
Can someone help me with this?

1.

### Settings

- Software version: ``?
- Battery used: `` 64 KW Ampera E/ Chevrolet Bolt
- Inverter communication protocol: `` ?
- Hardware used for Battery-Emulator: `HW_LILYGO,`
- CONTACTOR_CONTROL: `yes/
- DUAL_CAN: `yes/no`?
- WEBSERVER: `yes?
- MQTT: [`yes/no`?]([url](url))

dalathegreat, 2024-10-10:
Hi, see this wiki page for more info on how to get started with a new integration. https://github.com/dalathegreat/Battery-Emulator/wiki/Data-needed-for-new-battery-integration

It is unfortunately a quite long process, if you are starting on a vehicle type that previously has no CAN decoding available.

HVberg, 2024-10-15:
Hi Dala,
I can make Canlog data with the Lilygo now, from the car and from the stationaire battery, but how can I decodering the CAN data? Can you give me some advice with this maybe? 
Thanks

PR #642 (2024-12-23): Feature: Add base for Bolt-Ampera battery
### What
This PR implements the base needed for Bolt-Ampera batteries

### Why
To add support for more batteries

### How
There are still some TODOs in the .cpp file, that can be fixed once someone starts to test this battery type more!

Fixes #527
closed 2024-12-23 by dalathegreat