dalathegreat/Battery-Emulator #529: Battery not recognised by Fronius Gen24

GraphicObession, 2024-10-11:
Hi 

Great Project, thanks Dala!

I am setting up a BYD ATT 3 battery and a Fronius Gen24 ( not the plus version, but I have purchased a licence to do the upgrade, however I am not 100% sure they have installed the correct licence) and I am having a few issues as listed below:

1 - The Emulator appears to be communicating with the battery however the cell information is not loading 
2 - The Fronius has not correctly discovered the battery and I am unable to find any battery settings.

I have not connected the HV yet, would this have any affect on the above issues?

Any help is appreciated.

- Software version: `7.4.1`
- Battery used: `BYD ATTO 3`
- Inverter communication protocol: `ModBus`
- Hardware used for Battery-Emulator: `HW_LILYGO`
- CONTACTOR_CONTROL: `no` not tested
- DUAL_CAN: `no`
- WEBSERVER: `yes`
- MQTT: `yes`

[image]
[image]
[image]
[image]
[image]

dalathegreat, 2024-10-11:
Hi,
- Yes, High voltage needs to be present on the Inverter for it to proceed with checking it.
- Fronius configuration is done via the "Solar.start" android application
- Contact Fronius if the + license is not applied.
- Cellmonitor view is not available for Byd Atto 3 yet. We only have min/max cellvoltage.

GraphicObession, 2024-10-11:
Thanks Dala, I appreciate the info!

GraphicObession, 2024-10-25:
My setup is working now. 

Thanks again Dala!

closed 2024-10-25 by GraphicObession