dalathegreat/Battery-Emulator #534: Feature Request: Temperature Safety Shutdown

beadon, 2024-10-14:
### Expected Behavior

For safe operation - include support for an optional(?) temperature probe, attached to a lead will detect the temperature of the system, and safely shutdown the system, reporting the reason for forced shutdown to logging systems.

### Actual Behavior

Note sure if a temperature sensor is included ?

### Steps to Reproduce the Problem

N/A

### Settings

N/A

Note:
Perhaps this is a viable mid-term solution, if the ESP32 supports a chip-level temperature sensor ?
https://docs.espressif.com/projects/esp-idf/en/stable/esp32s2/api-reference/peripherals/temp_sensor.html

nagyrobi, 2024-10-14:
The chip's built temp sensor is notoriously unreliable.
A (set of) DS18B20 would be more suitable because:
- can be cabled relatively long (10s of meters)
- can have multiple sensors on the same GPIO thus multiple zones of the system can be monitored

dalathegreat, 2024-10-14:
@beadon Which system are you referring to? 

Since we use EV batteries, they all contain many temperature measurements. All battery integrations have a Min and Max temperature that we monitor and act on. If the battery gets too cold, or too hot, we stop using the battery. As an additional safety, the BMS inside the EV pack can act on its own, and reduce the allowed charge/discharge accordingly. So there really is no benefit to add more temperature measurements to the system.

closed 2024-10-14 by beadon

beadon, 2024-10-15:
Right.

I was on this path because of the different environmental conditions that each ESP32 is going to face during operation in different enclosures, especially in regions known to get seasonably warm. Places like Arizona, USA regularly get over 100 deg F, and this doesn't account for additional sun-provided thermals!

I have read that the ESP32 does not have an auto-off feature to protect itself from damage during a high-temperature environment, and wanted to get ahead of a thermal problem in the controller itself.

I have now closed this feature because I understand that it does require board-specific support, and reading the chip's temperature alone is not sufficient.  This is a safety and operational consideration for some kind of "production" environment, it's better to *know* that you're overtemp rather than guess if the hardware is functioning properly - you'll just be able to solve a shutdown (or safety) problem faster.

Please close if you also agree with this approach.  Keep open if you think it warrants more discussion and/or board support.