dalathegreat/Battery-Emulator #538: Keeping a battery pack in the leaf and wiring up to HV of solos hybrid inverter

Rbridge1988, 2024-10-14:
Good afternoon,

I have a solid 3.6kw hybrid inverter with pylon tech batteries. Is it possible to connect the 6kw DC side to the mppt input and wire into the DRM rs485 input via a lilygo board as mentioned my 2016 30kw leaf without taking the pack out of the car,? My plan is to use some big connectors hidden under the bonnet and connect in while the car is home. 

Thanks

dalathegreat, 2024-10-14:
Possible? Most likely with a lot of effort. Recommended? Not at all. Very easy to break the vehicle doing it this way.

nagyrobi, 2024-10-14:
How about the CHAdeMO port in v2x mode? What's the issue with that?

Rbridge1988, 2024-10-14:
> How about the CHAdeMO port in v2x mode? What's the issue with that?

So I'd have to energise the contactors by starting the car with a chademo connector to bring the HVDC to life and the mppt of the inverter connected to the other end ? Connect into the canbus and rs485 with the lilygo board?

nagyrobi, 2024-10-14:
Not sure....

Rbridge1988, 2024-10-15:
Would I be better off buying a 1kw 12v to 230v inverter, shutting the grid off and running the 1kw inverter as "mains"  and using the 12v battery in the car

dalathegreat, 2024-10-15:
Sounds like a discussion that would be better suited to a discussion forum like https://secondlifestorage.com/index.php

I will close this Github issue. Good luck with the build!

closed 2024-10-15 by dalathegreat