dalathegreat/Battery-Emulator #548: CHAdeMO Vehicles, Battery

NJbubo, 2024-10-20:
Hi Daniel,

Is the CHAdeMO code working? Use these code anyone?
I tried but it doesn't work for me.

Thanks

nagyrobi, 2024-10-20:
Do you use/have a proper CHAdeMO plug??

NJbubo, 2024-10-20:
Yes, I have.

nagyrobi, 2024-10-20:
Great! Can you point us in the right direction where did you get one?

leomeno, 2024-11-04:
Hi,
I also am trying to run the CHAdeMO implementation using Olimex ESP32-EVB, Solax Hybrid inverter and Nissan Leaf. I still don't have a CHAdeMO plug, I've ordered some cheaper from Aliexpress, bud I've already made 3rd order, because the previous 2 weren't shipped so they returned the money. But I've got some unfinished still only control-level (without DC power lines) testbed, connected the inverter to the board and connected the vehicle and CAN signals just using some pins in the vehicle's CHAdeMO  port. I had some problems, so I've made some modifications to the CHADEMO-BATTERY.cpp mostly some debug messages and tried to make commisioning of the communication, but I started with the version 7.5 and now I'm testing the 7.7.dev version, I need to test some more..

lenvm, 2024-12-12:
As #621 was merged, I will close this issue.

closed 2024-12-12 by lenvm