dalathegreat/Battery-Emulator #553: BMW I3 Battery SOC problem

mbuhansen, 2024-10-22:
### Expected Behavior

BMW I3 battery display soc show 100, soc on battery is on 51%. Cant get inverter to charge battery full.
Switch back to old soc solution HWbatt_soc

### Actual Behavior

### Steps to Reproduce the Problem

1.
1.
1.

### Settings

- Software version: ``
- Battery used: `BMW I3 60AH`
- Inverter communication protocol: ` Kostal`
- Hardware used for Battery-Emulator: `HW_LILYGO,`
- CONTACTOR_CONTROL: `no`
- DUAL_CAN: `no`
- WEBSERVER: `yes`
- MQTT: `yes'

[image]

havrla, 2024-10-27:
It is dangerous to use HWbatt_soc, this value is absolute, it decreases from top to bottom with time, 0 = zero is deep battery discharge. 100% cannot be reached because the upper buffer and degradation are subtracted from 100%.

Is this condition with a new battery that you bought somewhere or was it caused by operation?

The BMS in the battery sometimes makes a mistake in determining the SOC and energy and shows negative values ​​and needs to be fully charged at 100% and sleeping 30 min.

More info please.

We have to figure out a better way to treat this condition, but I have little information for that.

Nice app to find out info about the battery, what is it called?

dalathegreat, 2024-10-27:
In the latest main we can easily compare different SOC values:

[image]

```
  datalayer_extended.bmwi3.SOC_raw = (battery_display_SOC * 50);
  datalayer_extended.bmwi3.SOC_dash = (battery_HVBatt_SOC * 10);
  datalayer_extended.bmwi3.SOC_OBD2 = battery_soc;
```

At the moment the SOC we use for the software is battery_display_SOC, taken from 0x432

mbuhansen, 2024-10-27:
@havrla this i long story with this battery. The battery is BMW I3 60 AH, when the battery was in car and it was charging the battery get malfunction. The guy i buy battery from get to BMW to make diagnostic, and it wiil be expensive to repair battery bacause of bad cell. The gauy buy other used battery and put in the car. He try to open the bad battery, and found cell 96 was bad, he buy other singel cell an change only one cell and not the modul. it was not easy to dismantle but does it and all look fine.
i think the BMS calculate correct after this, the battery has charge to 100% i got it, and have try to restart battery both on low soc an high soc, but not deep sleep in 30 min.
How do you get the battry in deep slep, is it just to turn off lilygo and wakeup signal. or do i need to disconnect all power from battery...??

I cant understand why bms always say  100% when cell voltage only is 3.925 volt, can see other in discord forum can charge to 4.1

Do you now about is is possible to reset BMW BMS ISTA...???

The app i am using is a car diagnostic system Launch X-431 software is runing on tablet with bluetooth OBD dongle,

mbuhansen, 2024-10-27:
@dalathegreat Yes the new menu is fine , but i get it work when i use kostal inverter. If you can help to pull the new main in kostal brance i can test with new menu.

havrla, 2024-10-27:
@mbuhansen 
So you have a battery that was broken, someone probably repaired it, replaced one cell (it's impossible for me).

I had many batteries here, the article was never defective, always electronics.

99.7 does not equal 100% you must charge to 100%

There are several ways to explain this
What is the maximum voltage of the cell you observe when it is charging? Isn't it 4.07 V (60Ah)? and then the voltage should drop?

The difference between SOC display and raw can be real.
0% display = 10% raw
100% display = 100% raw minus 15% buffer minus degradation
51.9% raw and 100% display with approx. 33% degradation.

Your diagnostics also shows Battery charging 6528%, it will be about 65%
65% raw and 100% display is about 20% degradation.

And keep in mind that these exact calculations are based on an imprecise basis, because the battery when it was new had about 66 - 70Ah and some calculations and values ​​refer to the value when the battery was new, others to the label value of 60Ah.
So the interpretation suffers from a fundamental flaw of up to 10%

Battery management has 2 charge status indicators. The first in [%] the second in [kWh], these two indicators are linked to each other, but I work separately. When desynchronization occurs, what you describe can happen, the battery shows 99.5%, but only half the energy is needed, or vice versa. This usually happens with a car if the battery has been in storage for a long time, or it is somehow improperly communicated with, or the function is not terminated in an appropriate way. It can be fixed so that the car is charged to 100%, the car is put to sleep and the battery wakes up after about an hour. Then it will probably happen that according to the lowest voltage of the article, [%] is determined, it is converted to [kWh] and everything works - that is, a kind of calibration of the SW energy counter. If the car is driving, the battery is charged/discharged, the cell voltage is not checked (only the limit states are checked) and the vehicle's range is calculated from the energy value [kWh] and the percentage is calculated online somehow, but it does not have to match the value of the ratio [kWh] exactly .
As for sleep - I don't know how it works with the emulator, I haven't tried it yet. But in the car, when starting, each control unit receives the current time, so the BMS knows how long it has been sleeping and whether the battery voltage is at rest.

Try to check the reality, discharge the battery to 0% display, check if the balancing of the article is in order. Charge the battery to 100% display and measure the energy.

Your diagnosis shows SOH degradation, a fundamental deficiency.

However, for safe use of the battery, the SOC-display should be used as shown on the car's display. The remaining SOC values ​​are diagnostic values ​​that, if not interpreted correctly, can cause a fatal failure. For example, it is good to think that you will never reach 100%, or fixate on some value (thinking that it is 80% charged) - this is not good, that value will change over time as the battery degrades.

Try moving (charging/discharging) the battery and brag about the new measurement.

mbuhansen, 2024-10-28:
Thank you, good description you made.
About the cell voltage i have not the single voltage on cell, but battery max voltage when 100 in soc was 382 volt, and 7mv in delta voltage.

about the calibration procedure i will try, but i am not shure about Emulator send time value like the does. @dalathegreat do you now about lilygo send time to BMW battery on start up like havrla descripe above...?? if emulator not does i think i will be hard to make calibration.
I will try it out tommorrow, after battry has charge in the night.

i understand you when you say only electronic fails in battery, but it looks like bms is working on all cells, and i have see the old cell it was growing in size, and was bad.
[image]

havrla, 2024-10-28:
I'll start with your chart.
A drop in cell voltage after charging is normal. It takes approximately 15-60 minutes to charge the cell to its resting state.
So the fact that the voltage of the cell dropped after the end of charging is fine. It's just a matter of how much it will drop... if it's not too bad and the article is already soft.
It's better to compare several articles, you probably did it and it's just lost in translation when you show one piece, so the others didn't drop so much??

I will return to the BMS calibration, today I checked approximately 15 pieces of the battery that I bought, different pieces. I came across an exemplary pointer.

[image]

Look at the picture:
SOC display 0%
SOC HV  7%
Voltage 371V   is not 0% is not 7% !!

Here the BMS is decalibrated. The battery was left somewhere for a long time and lost its memory, or it was turned off incorrectly and did not save the state of the cell when it was turned off.
And as I said, this can be solved in a car so that the battery is charged to 100%, if the discrepancy is only a few percent, it will be calibrated immediately after charging. If it splits them hugely, for example here, it is necessary to turn off the battery/car, let it sleep and wait for the BMS to declare the cells to be calm, and to determine a new SOC value from the voltage.

havrla, 2024-10-31:
We tested version 7.5.0 today
The SOC_HW value is used, as requested by "mbuhansen".
Tested with goodwe.
I found that Goodwe will not discharge the battery below SOC 10%.

Here we have 2 worlds:
1. SOC_HW
0% is the destruction of the battery, 10% is a safe minimum, the maximum in % does not exist, or rather it is floating and decreases, because degradation is subtracted from SOCmax (example 100% minus 15% degradation, then SOCmax is 85% and will never be higher and this value will further decrease). SOCmax stops when the maximum voltage of the most charged cell is reached, for example 392-393V pack / 4.07V cell for 60Ah
SOC scaling cannot be done in this world because the top tier is uncertain.
"Real remaining capacity" is not adjacent to SOC_HW and its scaling is then meaningless.

2. SOC_Display
0% is a safe minimum, 100% is a fully charged battery when the maximum voltage is reached. "Real remaining capacity" is usable capacity and is adjacent to SOC_Display.

I think SOC_Display should be used as it is displayed on the vehicle's dashboard. The diagnostic value of residual energy should not be used, which we cannot adequately predict and interpret.

Does anyone have an opinion? If yes, please write the reason here.
Has anyone recently encountered the fact that their BMS would become uncalibrated during operation? Or is this the first case of a battery after repair?
I can imagine that over time it can lead to a wrong measurement of "Real remaining capacity" and "total used capacity". But we will probably have to wait another 1/2 year for that answer.

If someone has an uncalibrated BMS in the battery, it is not a reason for the entire project to suffer from reduced safety. We have to find another safe way to solve it, but we will find out soon, and "mbuhansen" will write more information with his recalibrated battery.

**Please use SOC_Display until the next version.**

mbuhansen, 2024-11-01:
I have not tested the new 7.5 on my setup yet because of my kostal inverter, the new battery info menu is not merge to kostal brance yet.
But when i look in the in v 7.5 BMW I3.cpp SOC dispaly s still used.
Dala wont change it as i request.

[image]

@havrla 
A comment to the battery calibration, the procedure you write. dont work with the emulator, i charge to 100 % and let battery sleep 1 hour, but battery don't recalibrate.
After this is use other procedure, i change bmw i3 cpp, to use SOC_HW. I charge battery full when i watch alle voltage in battery, 1 time i get implausible soc from bmw BMS, i restart battery and begin charging again. I stop charging when max cell voltage reach 4,1 volt. And switch back to use SOC_DISPLAY. 
Now i can charge 15 Kwh to battery 5 kwh more than before, so battery has calibrate now.
I think I one more time will do the procedure, and charge to battery bms goes down to zero in charge current, and see how much it change in the capacity.

SO yes it is best to use SOC_DISPLAY, but can be necessary  to switch to SOC_HW if battery not can full charge like my situation.

havrla, 2024-11-04:
my small mistake, i testing 7.6dev

havrla, 2024-11-04:
@mbuhansen       theoreticaly, is possible rescalind max 120% ?  :-)  try it...

mbuhansen, 2024-11-04:
@havrla Thanks maybe. But now it looks like i run okay, i can charge battery 16 KWH when using dispaly_soc. I only change to HW_soc if i need to push battery up, i can see when i charge with HW_soc and reach about 82% battery bms only want to charge with 1 kwh, so bms is derate charge power. but have not test when bms set derate to zero amp.

havrla, 2024-11-20:
If the BMS says not to charge, then it shouldn’t be charged—either the maximum voltage has been reached, or a safety limitation is in place. As I mentioned, degradation is subtracted from SOC_HW from the top, so you won’t get 100%, and it will continue to decline over time.

havrla, 2024-11-20:
Yesterday, a gentleman from Switzerland brought me a 120Ah battery for repair. This battery is very interesting and worth detailing here. The battery has an unknown history, but inside it contains two new modules and six original old ones. The battery reports a range of 50 km, and the energy between 0-100% is about 10 kWh. In diagnostics, the battery indicates that cell module number 8 is damaged.
[image]
Here you can see SOC_Display at 100% and SOC_HW at 25.3%—indicating the battery is in emergency mode.
[image]
Using a special tool, it is possible to extract information from the BMS about its assessment of the individual modules' condition. Module number 8 has a capacity of 78%, while the new modules are at 100%.
[image]
It also considers that the module with the highest capacity has 128Ah, while the module with the lowest capacity has 97Ah. That’s a difference of 31Ah!

It’s important to understand that if an issue arises with a battery module, the accumulator may enter emergency mode. Relying on SOC_HW and attempting to charge the accumulator in this state is highly unwise.

Repair instructions: For 60Ah and 94Ah batteries, it is sufficient to replace the module and perform calibration. However, for 120Ah batteries, an EoS test is strictly required.
[image]
[image]
[image]

mbuhansen, 2024-11-20:
Thank you, very god description how to repair the battery correct. But i think deep sleep function in car to initialise the correct display soc is not work with the emulator maybe because emulator not send time to SME. 
My battery has be good for the last 3 weeks i can charge 15-16 kwh to the battery now.

havrla, 2025-02-21:
By the way, I wanted to say that for the 120Ah version, if the BMS evaluates an error, it will display 100% = 20%RAW. Not possible charging to 100%.

dalathegreat, 2025-06-10:
Please create a new issue for your Foxess @sprocketman