dalathegreat/Battery-Emulator #554: Mismatch of SOC values ZOE battery GEN1

valentzoe, 2024-10-22:
### Expected Behavior

The SOC calculation requires a significant correction for Battery ZOE GEN1.

### Actual Behavior

I noticed that the SOC calculated by the emulator is:
-70% and the real SOC is 93%,
-20% and the real SOC is 55%, etc.
On the photo: charge-discharge curves of real Li-Ion, NMC 36 Ah.

[image]

[image]

- Software version: `7.5`
- Battery used: `Renault ZOE GEN1`
- Inverter communication protocol: `CAN`
- Hardware used for Battery-Emulator: `HW_LILYGO`
- CONTACTOR_CONTROL: `yes`
- DUAL_CAN: `no`
- WEBSERVER: `yes`
- MQTT: `no`

dalathegreat, 2024-10-30:
We use the SOC% sent by battery, does this mean that the battery itself is not sending the correct SOC% when discharging?

valentzoe, 2024-10-31:
Hi! Probably not. It probably sends this data to the ECU ZOE. Here is a picture.
[image]
[image]

valentzoe, 2024-10-31:
[image]

dalathegreat, 2024-10-31:
What is the lowest you can discharge the battery to?

valentzoe, 2024-10-31:
Did as you requested today and discharged it until it was completely cut off by the inverter. And this is what I got.
[image]

valentzoe, 2024-10-31:
I discharged the battery at your request. And this is what I got:
[image]

dalathegreat, 2024-11-04:
Okay so at 20% SOC left the voltage was 3600mV/cell. This sounds like quite common thing on EVs, on the Nissan LEAF it is not normal to go under 3500mV/cell, there is barely any capacity left there

[image]

I suggest removing the "[Rescale SOC%](https://github.com/dalathegreat/Battery-Emulator/wiki/Webserver-guide#rescale-soc)" setting, and trying to use the battery all the way down to 0-5%. This should correspond with anywhere between 3000-3500mV. Do note that the furthest discharge point is also based on temperate, and SOH%. I really dont think we have any bug here.

closed 2025-01-04 by dalathegreat