dalathegreat/Battery-Emulator #564: Tesla Model 3 battery, 2021 RWD, and Deye SUN-50k-SG01HP3-EU-BM4 inverter – the inverter does not detect the battery capacity.

XMLNem, 2024-10-27:
### Expected Behavio
Tesla Model 3 battery, 2021 RWD, and Deye SUN-50k-SG01HP3-EU-BM4 inverter – the inverter does not detect the battery capacity.

I started the battery using a LilyGO ESP; the battery starts up, and information such as voltage, status, etc., is displayed on the web interface. However, the inverter does not detect the battery capacity. Please help.
### Actual Behavior
[image]
[image]

### Steps to Reproduce the Problem

### Settings

- Software version: 6.4.0
- Battery used: Tesla Model 3 battery, 2021 RWD 1104428-00-W
- Inverter communication protocol: Lithium Mode: 01
- Hardware used for Battery-Emulator: `HW_LILYGO`
- CONTACTOR_CONTROL: `no`
- DUAL_CAN: `yes`
- WEBSERVER: `yes`
- MQTT: `no`

XMLNem, 2024-10-27:
#ifndef __USER_SETTINGS_H__
#define __USER_SETTINGS_H__
#include <stdint.h>

/* This file contains all the battery/inverter protocol settings Battery-Emulator software */
/* To switch between batteries/inverters, uncomment a line to enable, comment out to disable. */
/* There are also some options for battery limits and extra functionality */
/* To edit battery specific limits, see also the USER_SETTINGS.cpp file*/

/* Select battery used */
//#define BMW_I3_BATTERY
//#define BYD_ATTO_3_BATTERY
//#define CHADEMO_BATTERY	//NOTE: inherently enables CONTACTOR_CONTROL below
//#define IMIEV_CZERO_ION_BATTERY
//#define JAGUAR_IPACE_BATTERY
//#define KIA_HYUNDAI_64_BATTERY
//#define KIA_E_GMP_BATTERY
//#define KIA_HYUNDAI_HYBRID_BATTERY
//#define MG_5_BATTERY
//#define NISSAN_LEAF_BATTERY
//#define PYLON_BATTERY
//#define RENAULT_KANGOO_BATTERY
//#define RENAULT_ZOE_GEN1_BATTERY
//#define RENAULT_ZOE_GEN2_BATTERY
//#define SANTA_FE_PHEV_BATTERY
#define TESLA_MODEL_3_BATTERY
//#define VOLVO_SPA_BATTERY
//#define TEST_FAKE_BATTERY

/* Select inverter communication protocol. See Wiki for which to use with your inverter: https://github.com/dalathegreat/BYD-Battery-Emulator-For-Gen24/wiki */
#define BYD_CAN          //Enable this line to emulate a "BYD Battery-Box Premium HVS" over CAN Bus
//#define BYD_MODBUS  //Enable this line to emulate a "BYD 11kWh HVM battery" over Modbus RTU
//#define LUNA2000_MODBUS  //Enable this line to emulate a "Luna2000 battery" over Modbus RTU
//#define PYLON_CAN        //Enable this line to emulate a "Pylontech battery" over CAN bus
//#define SMA_CAN          //Enable this line to emulate a "BYD Battery-Box H 8.9kWh, 7 mod" over CAN bus
//#define SMA_TRIPOWER_CAN //Enable this line to emulate a "SMA Home Storage battery" over CAN bus
//#define SOFAR_CAN        //Enable this line to emulate a "Sofar Energy Storage Inverter High Voltage BMS General Protocol (Extended Frame)" over CAN bus
//#define SOLAX_CAN        //Enable this line to emulate a "SolaX Triple Power LFP" over CAN bus

/* Select hardware used for Battery-Emulator */
#define HW_LILYGO
//#define HW_STARK

/* Other options */
//#define DEBUG_VIA_USB  //Enable this line to have the USB port output serial diagnostic data while program runs (WARNING, raises CPU load, do not use for production)
//#define DEBUG_CANFD_DATA    //Enable this line to have the USB port output CAN-FD data while program runs (WARNING, raises CPU load, do not use for production)
//#define INTERLOCK_REQUIRED  //Nissan LEAF specific setting, if enabled requires both high voltage conenctors to be seated before starting
//#define CONTACTOR_CONTROL     //Enable this line to have pins 25,32,33 handle automatic precharge/contactor+/contactor- closing sequence
//#define PWM_CONTACTOR_CONTROL //Enable this line to use PWM logic for contactors, which lower power consumption and heat generation
//#define DUAL_CAN              //Enable this line to activate an isolated secondary CAN Bus using add-on MCP2515 controller (Needed for FoxESS inverters)
//#define CAN_FD  //Enable this line to activate an isolated secondary CAN-FD bus using add-on MCP2517FD controller (Needed for some batteries)
//#define SERIAL_LINK_RECEIVER  //Enable this line to receive battery data over RS485 pins from another Lilygo (This LilyGo interfaces with inverter)
//#define SERIAL_LINK_TRANSMITTER  //Enable this line to send battery data over RS485 pins to another Lilygo (This LilyGo interfaces with battery)
#define WEBSERVER  //Enable this line to enable WiFi, and to run the webserver. See USER_SETTINGS.cpp for the Wifi settings.
#define LOAD_SAVED_SETTINGS_ON_BOOT  //Enable this line to read settings stored via the webserver on boot (overrides Wifi/battery settings set below)
//#define FUNCTION_TIME_MEASUREMENT  // Enable this to record execution times and present them in the web UI (WARNING, raises CPU load, do not use for production)

/* MQTT options */
// #define MQTT  // Enable this line to enable MQTT
#define MQTT_SERVER "192.168.xxx.yyy"
#define MQTT_PORT 1883

/* Event options*/
#define DUMMY_EVENT_ENABLED false  //Enable this line to have a dummy event that gets logged to test the interface

/* Select charger used (Optional) */
//#define CHEVYVOLT_CHARGER //Enable this line to control a Chevrolet Volt charger connected to battery - for example, when generator charging or using an inverter without a charging function.
//#define NISSANLEAF_CHARGER //Enable this line to control a Nissan LEAF PDM connected to battery - for example, when generator charging

/* Battery settings */

// Predefined total energy capacity of the battery in Watt-hours
#define BATTERY_WH_MAX 55000
// Increases battery life. If true will rescale SOC between the configured min/max-percentage
#define BATTERY_USE_SCALED_SOC true
// 8000 = 80.0% , Max percentage the battery will charge to (Inverter gets 100% when reached)
#define BATTERY_MAXPERCENTAGE 800
// 2000 = 20.0% , Min percentage the battery will discharge to (Inverter gets 0% when reached)
#define BATTERY_MINPERCENTAGE 200
// 300 = 30.0A , BYD CAN specific setting, Max charge in Amp (Some inverters needs to be limited)
#define BATTERY_MAX_CHARGE_AMP 300
// 300 = 30.0A , BYD CAN specific setting, Max discharge in Amp (Some inverters needs to be limited)
#define BATTERY_MAX_DISCHARGE_AMP 300

extern volatile uint8_t AccessPointEnabled;
extern const uint8_t wifi_channel;

/* Charger limits (Optional): Set in the USER_SETTINGS.cpp or later in the webserver */
extern volatile float charger_setpoint_HV_VDC;
extern volatile float charger_setpoint_HV_IDC;
extern volatile float charger_setpoint_HV_IDC_END;
extern volatile float CHARGER_SET_HV;
extern volatile float CHARGER_MAX_HV;
extern volatile float CHARGER_MIN_HV;
extern volatile float CHARGER_MAX_POWER;
extern volatile float CHARGER_MAX_A;
extern volatile float CHARGER_END_A;
extern bool charger_HV_enabled;
extern bool charger_aux12V_enabled;

#endif

Gangster8214, 2024-10-29:
Hello

This error is actual for my SUN-50K too.
The problem occurs on all HV DEYE inverters. 5K-50K.

I noticed that the Battery capacity is always equal to the Charge current limit. 

[image]

NemesisXam, 2024-10-29:
> Hello
> 
> This error is actual for my SUN-50K too. The problem occurs on all HV DEYE inverters. 5K-50K.
> 
> I noticed that the Battery capacity is always equal to the Charge current limit.
> 
> ![изображение_viber_2024-10-29_10-34-11-329](https://private-user-images.githubusercontent.com/116462989/381044517-5a70da41-8745-46d1-a0be-17c23fa7963d.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzAyMDIwOTMsIm5iZiI6MTczMDIwMTc5MywicGF0aCI6Ii8xMTY0NjI5ODkvMzgxMDQ0NTE3LTVhNzBkYTQxLTg3NDUtNDZkMS1hMGJlLTE3YzIzZmE3OTYzZC5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMDI5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTAyOVQxMTM2MzNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zNzcwMDhmYjY4YzBhYjE1ZDQ0MjBiZTIyNjY0NTAwMmY5YTRiNTVkNmJmY2Y4Nzg3YzZkNTIzOTI0NzE2ZmQxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.6ZYdbt69_6zAMSH-Yr1KBzMTyrNXP1XcS7BKisjdWpM)

Can I view your battery settings on the LilyGO?

dalathegreat, 2024-10-29:
This is a clear bug, most likely due to https://github.com/dalathegreat/Battery-Emulator/issues/562

Should be safe to continue using until we have a fix

NemesisXam, 2024-10-29:
> This is a clear bug, most likely due to #562
> 
> Should be safe to continue using until we have a fix

Thank you. How will the Deye know when to charge the battery?

dalathegreat, 2024-10-29:
Fix for this is now available in this pull request https://github.com/dalathegreat/Battery-Emulator/pull/568

@NemesisXam This only affects capacity Ah number, not used for deciding when to charge. It is a bug that only affects a visual value.

mace-de, 2024-10-29:
> > This is a clear bug, most likely due to #562
> > Should be safe to continue using until we have a fix
> 
> Thank you. How will the Deye know when to charge the battery?

Deye charges the battery based on the SOC and allowed charge current transmitted by the BMS.

NemesisXam, 2024-10-29:
> Fix for this is now available in this pull request #568
> 
> @NemesisXam This only affects capacity Ah number, not used for deciding when to charge. It is a bug that only affects a visual value.

I understand that it's necessary to update the firmware?

Gangster8214, 2024-10-29:
> Can I view your battery settings on the LilyGO?

Hello

Now I use this settings

[image]

dalathegreat, 2024-10-29:
Please try the PR568 with an attempted fix: 
[Battery-Emulator PR568.zip](https://github.com/user-attachments/files/17559925/Battery-Emulator.PR568.zip)

XMLNem, 2024-10-29:
> Будь ласка, спробуйте PR568 зі спробою виправлення: [Battery-Emulator PR568.zip](https://github.com/user-attachments/files/17559925/Battery-Emulator.PR568.zip)

Alright, the result will be around this time tomorrow. And is an OTA update possible?

dalathegreat, 2024-10-30:
Confirmed workon on the Discord:

[image]

Pull request is merged (now in main 7.6.dev), this fix will be included in 7.6.0 tag

closed 2024-10-30 by dalathegreat