dalathegreat/Battery-Emulator #565: Tesla battery contactor close control before safety is checked

frasersimpson83, 2024-10-28:
### Expected Behavior
Just when discussing this with other users in the Tesla thread, we realised that the emulator code will just send the required CAN commands to close contactors regardless of whether or not the safety boundries in the software are met.

eg cell deviation too high or low, cell voltage too high or low etc.

We thought that maybe it would make more sense for the Emulator to sit in a 'checking' state to wait for cell voltages (I know this can take a minute) and once it's checked all that to make sure it's within bounds then it can start sending CAN packets to close contactors.

That seems to make more sense.    As long as it shows on the web interface it's just waiting to check out data before starting battery contactor close that seems to be safer in my mind and I assume not too hard to implement as it's all there just flipping it around to check before sending CAN instead of after.

josiahhiggs, 2024-11-27:
I'm happy to have a look into this. @dalathegreat, do you have any examples of where this code is best to modify?

dalathegreat, 2024-11-27:
@josiahhiggs , easy fix, just add a delay on the start of the send_can function, so that the function returns right away if not enough time has passed 

Somewhere here: https://github.com/dalathegreat/Battery-Emulator/blob/main/Software/src/battery/TESLA-BATTERY.cpp#L1199

josiahhiggs, 2024-12-26:
I have attempted to make the recommended changes and PR 702 is now up for review.