dalathegreat/Battery-Emulator #569: Need help with custom battery, CAN  list available

MathieuWuyts, 2024-10-29:
Hello,

I need help implementing a custom battery. I have a complete CAN list available. Precharge and contactors all programmed within the battery itself, so only needs a closed contact and 12V source to turn on.
I checked the existing .h and .cpp files of a few batteries, but I'm not sure how I can adapt my battery to the existing ones.
Anyone willing to help?

thanks

dalathegreat, 2024-10-29:
@MathieuWuyts 
What battery manufacturer? What Model? Datasheets? CAN list?

Please attach more information here

MathieuWuyts, 2024-10-29:
Hi Dala, it is a custom battery by Intercel in the Netherlands. Can list in attachment.
[CAN_Messages_Cellpower_CLPL_Battery.pdf](https://github.com/user-attachments/files/17557457/CAN_Messages_Cellpower_CLPL_Battery.pdf)

dalathegreat, 2024-10-30:
PR is now open! https://github.com/dalathegreat/Battery-Emulator/pull/575

closed 2024-11-04 by dalathegreat