dalathegreat/Battery-Emulator #572: Double Battery Automatic Contactor Support GPIO

malcolmputer, 2024-10-30:
### Expected Behavior

It would be great for double battery support to also support automatic contactor support utilizing GPIO, something like this: 

// Contactor handling
#define POSITIVE_CONTACTOR_PIN 32
#define NEGATIVE_CONTACTOR_PIN 33
#define PRECHARGE_PIN 25

#define 2ND_POSITIVE_CONTACTOR_PIN 13
#define 2ND_NEGATIVE_CONTACTOR_PIN 16
#define 2ND_PRECHARGE_PIN 18

### Actual Behavior

Currently this has to be done manually, or utilizing a battery that supports contactor control over CAN, since the 3lb board that's in development supports multiple relays, it makes sense to support this.

dalathegreat, 2024-11-15:
PR for this is UP :raised_hands: https://github.com/dalathegreat/Battery-Emulator/pull/619

dalathegreat, 2024-11-24:
Closed as completed! Target for 7.8.0 release!

closed 2024-11-24 by dalathegreat