dalathegreat/Battery-Emulator #578: BMW I3    WUP signal   Pin25

havrla, 2024-10-31:
### Expected Behavior

version 7,5,0

It does not fully control pin25 to activate WakeUP.

Pin25 is programmed to turn on pin25 after start-up and keep it on all the time.

However, the web interface has a "contactors open"/"contactors close" button. Sigal WUP must also reflect this button. Off works but on doesn't, WUP has to go down and up. We tested by rough programming and then it worked. Contactors open -> WUp down, Contactors connected -> WUp up.

Please add a function for controlling WUP - pin25

Thanks 
Havrla

dalathegreat, 2024-11-03:
Thanks for the report @havrla , fix is up in this PR: https://github.com/dalathegreat/Battery-Emulator/pull/585

dalathegreat, 2024-11-03:
Fix is merged to main, feel free to test it if you have time!

closed 2024-11-04 by dalathegreat