dalathegreat/Battery-Emulator #587: Add NVROL and temporisation to Zoe Ph2

dalathegreat, 2024-11-04:
### Expected Behavior

### Actual Behavior

### Steps to Reproduce the Problem

1.
1.
1.

### Settings

- Software version: ``
- Battery used: ``
- Inverter communication protocol: ``
- Hardware used for Battery-Emulator: `HW_LILYGO, HW_STARK, Custom`
- CONTACTOR_CONTROL: `yes/no`
- DUAL_CAN: `yes/no`
- WEBSERVER: `yes/no`
- MQTT: `yes/no`

mypi-home, 2025-01-19:
I have developed some of my own code for this and testing over the next couple of weeks.  If there is no-one else working on this I can potentially contribute it back to BE if it works for others to try too.

Basically need to send an extra frame in addition to 373 synced to time and send extra frames on shut down.

ianjenn, 2025-02-13:
Did you manage to overcome the time confusion state. I was about to purchase a gen 2 battery but hesitating until this issue fixed.

Kasper-Kristensen, 2025-03-13:
How do you perform the NVROL reset?

I'm trying to get a Fronius Gen24 10.0 plus to run with a Renault Zoe 52kWh battery. 
The voltage is fine (around 350V DC), but I do not receive any canbus communication. 

I already have another system running with a Fronius Gen24 5.0 plus with a Nissan Leaf 24kWh battery.

[image]

kyberias, 2025-03-23:
Curious about your code for this issue, @mypi-home ... is it in a state that it could be contributed to BE? Probably there are people that could help or support with that.

lenvm, 2025-03-27:
Has anyone already tried the following (pseudocode)?

NVROL Reset:
```
cansend 18DADBF1#021003AAAAAAAAAA
sleep 0.1
cansend 18DADBF1#043101B00900AAAA
```

Enable temporisation before sleep: 
```
cansend 18DADBF1#021003AAAAAAAAAA
sleep 0.1
cansend 18DADBF1#042E928101AAAAAA"
```

Source: [ZE50 Reset Commands](https://github.com/ljames28/Renault-Zoe-PH2-ZE50-Canbus-LBC-Information?tab=readme-ov-file#commands)

mypi-home, 2025-04-05:
Yes I have tried this and also implementing additional 376 time frame but this has not resolved the issue. The BMS is still in a panic state and lots of the values it is reporting are incorrect. Are you on the discord to discuss this further? I just posted my findings.

ljames28, 2025-04-19:
I've done some testing on which frames are mandatory. You need to be sending frame 373 and 376 every 100ms. Below is the Python code I'm using. After the NVROL reset, the battery happily balances. There is detailed info about what the frames do in the excel file in the repo here https://github.com/ljames28/Renault-Zoe-PH2-ZE50-Canbus-LBC-Information

```
            Seconds_since_production = time.time() - 1614454107
            Minutes_since_production = Seconds_since_production / 60
            Year_unfloored = Minutes_since_production / 255 / 255
            Year_seg = math.floor(Year_unfloored)
            Remainder_years = Year_unfloored - Year_seg
            Remainder_hours_unfloored = (Remainder_years * 255)
            Hour_seg = math.floor(Remainder_hours_unfloored)
            Remainder_hours = Remainder_hours_unfloored - Hour_seg
            Minutes_seg = math.floor(Remainder_hours * 255)

 Payload_373 = [0xC1, 0x40, 0x5D, 0xB2, 0x0, 0x1, 0xff, 0xe3]
 Payload_376 = [Year_seg, Hour_seg, Minutes_seg, Year_seg, Hour_seg, Minutes_seg, 0xA, 0x0]
```

PR #1017 (2025-05-06): Renault Zoe Gen2: add NVROL reset function
### What

#### Aim 🚀
This PR implements the NVROL reset function described on [ljames28/Renault-Zoe-PH2-ZE50](https://github.com/ljames28/Renault-Zoe-PH2-ZE50-Canbus-LBC-Information?tab=readme-ov-file#commands), and as is mentioned on #587.

#### Testing ⚠️⚡Testers welcome ⚡ ⚠️
This PR shall be checked by someone that has a Renault Zoe Gen2 battery.

Fixes #587

### How
By implementing a new function `transmit_reset_nvrol_frames()` that implements the following pseudo code:

```
cansend 18DADBF1#021003AAAAAAAAAA
sleep 0.1
cansend 18DADBF1#043101B00900 [... cut]
closed 2025-05-06 by lenvm