dalathegreat/Battery-Emulator #592: LG Resu 10H/16H prime support

ks123k, 2024-11-04:
Hi, does anyone have a log of CAN communication between SMA inverter and battery? I'm looking for log using [this](https://github.com/dalathegreat/Battery-Emulator/blob/main/Software/src/inverter/SMA-CAN.cpp) protocol.

I would like to add support for a resu prime batteries, it looks like the battery is using the SMA protocol mentioned above but I need to emulate the inverter for the battery to enable it's output.

Unfortunately I don't have access to compatible inverter.

dalathegreat, 2024-11-05:
Yes, we have CAN logs. @malcolmputer over on the Discord was able to log it. We even added a PR in progress for adding LG Resu SMA support: https://github.com/dalathegreat/Battery-Emulator/pull/398

In the end we closed this branch, and went with the SMA-BYD can instead: https://github.com/dalathegreat/Battery-Emulator/pull/410

The technical reason behind this decision, was due to the SMA-BYD being simpler to work with. The RESU battery is a low voltage battery, with an internal DC-DC converter to boost the 48V up to high voltage. We figured it would be a total pain to emulate this DC-DC converter, so we just went for SMA-BYD.

Is there some reason you cannot use SMA-BYD?

ks123k, 2024-11-05:
@dalathegreat  I have two of these batteries laying around, I got them for 1/5 of the regular price and the resu app claims they have 100% SoH. I would like to use them with sofar and I'm wondering if it's even possible or the inverter needs to send a message "hey I'm going to discharge you" / "hey I'm going to charge you" so that the dc-dc converter can switch modes or does it somehow work automatically. 

Is there a chance you could provide me actual log so I can see the order of CAN messages during startup or should I join the discord?

dalathegreat, 2024-11-05:
@ks123k Ah, so the other way around! This will be *extremely* difficult, since you need to control the DC/DC converter to generate a matching output voltage. Maybe malcolmputer can share the logs with you directly, the problem with them is that they contain serial numbers which we should not share freely (risk getting the person who took the logs in trouble, banning their S/N potentially)

Good luck with this, I think you will be the only person in the world to try and use the RESU batteries in "reverse"

dalathegreat, 2026-09-28:
Closing as obsolete, no movement on LG Resu

closed 2026-09-28 by dalathegreat

SkyJetty, 2026-11-12:
Hi @malcolmputer and @dalathegreat,

I'm trying to connect an LG RESU16H Prime high-voltage battery to an SMA Sunny Boy Smart Energy SBSE5.8-US-50 inverter running firmware 03.16.21.R.

The LG battery was commissioned using LG enblock Manager, and the CAN wiring appears correct, but the SMA (running firmware 03.16.21.R) reports event 9308 (Battery system communication error) and does not recognize the battery.  LG Firmware versions are BMS 2.6.3.4 and DC/DC 1.4.0.0.

I've been reviewing the experimental SMA-RESU implementation in PR #398 and the subsequent SMA/BYD implementation in PR #410.

Would you be willing to share a sanitized CAN capture from a working LG RESU Prime and SMA Sunny Boy Storage installation, particularly the initial handshake, battery identification, DC/DC converter activation, and shutdown sequences?

I understand the concerns about exposing battery serial numbers and don't need any identifying information.

I'm also interested in whether you've identified differences between the older SMA Sunny Boy Storage protocol and the newer Sunny Boy Smart Energy protocol.

Any documentation or experimental code would be greatly appreciated.

Thank you!